openhwfoundation/cv32e40p #636: Problem removing APU signals

moimfeld, 2021-04-19:
Hello,

I am doing an implementation of the x-interface on cv32e40p. Before I can implement the new interface I need to remove the APU. For removing all the APU and FPU signals I use Cadence Conformal v19.10.100 to check the logical equivalence of the changed (revised) core and the cv32e40p_v1.0.0 core (golden). To automate the equivalence check I am using [Davide Schiavones script](https://github.com/davideschiavone/cv32e40p/tree/lec_script/scripts/cadence_conformal). You can find the logically equivalent core with (almost) all APU/FPU signals removed [here](https://github.com/moimfeld/cv32e40p/tree/x-interface). 

## Issue
When removing apu_en_ex_o and all the signals derived from it Conformal finds that the golden and the revised core are not logically equivalent anymore.
Does anyone have an idea why one cannot remove these signals without changing the logical equivalence? If yes, what can be done to resolve this issue and remove the signals?
#### About the apu_en_ex_o signal:
- The signal originally came from the decoder but I was able to remove it there and keep the logical equivalence. So in my version of the core the signal starts in cv32e40p_id_stage.sv and gets passed on via the core module to the cv32e40p_ex_stage.sv.
- Since there is no more APU in the core anymore, the signal stays zero at all times.
- The signal cannot be hardwired to zero (e.g. assign apu_en_ex_o = ‘0) without Conformal reporting non logical equivalence. The signal therefore gets assigned to zero in the ID/EX pipeline registers (In the golden model the signal also gets assigned in the ID/EX pipeline registers).

## Attempts to resolve the problem
There is only one major difference in Cadences verification report between the logically equivalent core and the non logically equivalent core where I removed the apu_en_ex_o signal and the ones derived from that.
Verification Report for logically equivalent core:
`Not-mapped DFF/DLAT is detected:                          yes *`
Verification Report for non logically equivalent core:
`Not-mapped DFF/DLAT is detected:                          no`
Since this difference indicates that the assignment in the pipeline registers might be the reason for the non logical equivalence I searched for a solution in the manual. I found the command “set flatten model -seq_constant”. This command solved the issue but I was able to negate some other inputs of the pipeline registers without causing conformal to find non logical equivalence. I conclude that this command is not the solution to the problem, since it compromises the effectiveness of the LEC. I also tried other options of this command like -seq_constant_feedback and -seq_constant_x_to 0 without success.

## Report and Log Files
I uploaded one version of the logs/reports before removing the apu_en_ex_o signal, where the core is logically equivalent (EQ) and one version after removing the signal where the core is not logically equivalent (NEQ) anymore.
[EQ_result_verfication.txt](https://github.com/openhwgroup/cv32e40p/files/6335364/EQ_result_verfication.txt)
[EQ_result.txt](https://github.com/openhwgroup/cv32e40p/files/6335340/EQ_result.txt)
[EQ_cv32e40p_lec_log.log](https://github.com/openhwgroup/cv32e40p/files/6335409/EQ_cv32e40p_lec_log.log)
[NEQ_result_verfication.txt](https://github.com/openhwgroup/cv32e40p/files/6335352/NEQ_result_verfication.txt)
[NEQ_result.txt](https://github.com/openhwgroup/cv32e40p/files/6335350/NEQ_result.txt)
[NEQ_cv32e40p_lec_log.log](https://github.com/openhwgroup/cv32e40p/files/6335410/NEQ_cv32e40p_lec_log.log)
[NEQ_result_noneq.txt](https://github.com/openhwgroup/cv32e40p/files/6335351/NEQ_result_noneq.txt)

MikeOpenHWGroup, 2021-04-19:
Hi @moimfeld.  Perhaps you would be interested in the [cv32e40x](https://github.com/openhwgroup/cv32e40x)?  The User Manual needs updating, but the [project proposal](https://github.com/openhwgroup/core-v-docs/blob/master/program/Project%20Descriptions%20and%20Plans/CV32E40X/CV32E40X-PPL.md) _should_ give you a very good idea of what this core is all about.  If you are interested in a "CV32E40P with an X interface", this is a good place to start.

davideschiavone, 2021-04-19:
Thanks a lot @moimfeld , amazing report and progress done so far.

I have no answer to your question at the moment as I have seen the problem together with you. I ping more expert people in the field that may have an answer to your questions: @Silabs-ArjanB, @clairexen, @jquevremont  and @treforsouthwell

Hi @MikeOpenHWGroup , Moritz is implementing the X interface in the CV32E40P core as agreed to interface the FPU to that. The CV32E40X RTL will be inspired by his work.

treforsouthwell, 2021-04-19:
I'm not an expert in Conformal and it's hard to tell from this log file what it thinks is non-equivilent. It maybe a good idea to open LEC in the GUI and view the differences there.
If you get stuck I could run formality on the change, removing an output shouldn't really change a DFF.

moimfeld, 2021-04-19:
@Silabs-ArjanB Thank you for your suggestion. If I understand correctly you propose to disconnect the apu_en signal from the decoder to the id stage (and assign apu_en to '0). But I have already successfully done that [here](https://github.com/moimfeld/cv32e40p/blob/x-interface/rtl/cv32e40p_id_stage.sv), and the cores stayed logically equvalent.  The problem arises when I want to delete the signal **apu_en_ex_o** from the cv32e40p_id_stage.sv. When I directly assign apu_en_ex_o to '0 cadence conformal finds that the golden and the revised core are non logically equivalent. That is why I am assigning apu_en_ex_o to 0 in the pipeline registers in the id stage (I am assigning it in the pipeline registers because in the golden core (v1.0.0) the apu_en_ex_o signal also gets assigned in the pipeline registers).
I will check if I can prove with fromal technology that apu_en_ex_o = 1'b0 (even though I am also quite sure that for the frozen parameters thats the case).
I will probably stick to your second suggestion and add a comment to the lines I did not remove.
Still, if I can formally prove that apu_en_ex_o is stays 0 at all times, it would be important to make appropriate changes to the LEC script to get rid of "false negatives".

Silabs-ArjanB, 2021-04-20:
Hi @moimfeld ,

> That is why I am assigning apu_en_ex_o to 0 in the pipeline registers in the id stage (I am assigning it in the pipeline registers because in the golden core (v1.0.0) the apu_en_ex_o signal also gets assigned in the pipeline registers).

This is the right thing to do in my opinion. The reason you cannot remove it or add a simple assignment statement to 0 is because that signal is not actually always 0. If the core is not reset immediately this signal can remain undefined for many clock cycles I expect. So, for now I would be in favor to just keep that signal assigned to 0 in the pipeline registers as you did and add a comment why you did that (as opposed to somehow masking this point).