openhwfoundation/cv32e40p #639: Implementation Example

embedded-programmer, 2021-04-27:
Hi,

I've implemented Ibex core in FPGA using [https://github.com/lowRISC/ibex/tree/master/examples/fpga/artya7](url) successfully. Due to my need for floating-point-unit in the project, I switched the core from Ibex to cv32e40p. However, it didn't work.
* I just simply changed the core instantiation in top_arta7.sv from Ibex to cv32e40p. Did I miss something? Do you have any suggestions or examples on how to implement this core to FPGA?

MikeOpenHWGroup, 2021-04-27:
Hi @embedded-programmer, thanks for your interest.  The current version of the CV32E40P has not been verified with the floating point unit.  The current status of the CV32E40P is captured in its [User Manual on ReadTheDocs](https://cv32e40p.readthedocs.io/en/latest/intro/).  In particular you should read the [core integration](https://cv32e40p.readthedocs.io/en/latest/integration/) and [core versions](https://cv32e40p.readthedocs.io/en/latest/core_versions/) chapters.

The CV32E40P is descended from the PULP Platform RI5CY project that used "fpnew" project from PULP Platform (https://github.com/pulp-platform/fpnew) to implement the floating point ISA.  Fpnew is known to be stable, although not fully verified.

You can take one of three paths to get the CV32E40P working with fpnew:

1. Look at the "APU interface" specification of the CV32E40P.   I believe this interface is pin-compatible with fpnew.  Keep in mind that this interface does not function unless the top level parameter FPU in CV32E40P is set to 1.
2. There is an example on how to connect CV32E40P core and an FPU (FPNEW) in the CV32E40P repository.  Go to the example_tb/core folder and type: make fpnew_deps which clones the fpnew sources into the RTL folder.  The command make vsim-all-fp will compile it with Modelsim.
3. We are aware of an Academic group in the USA that is investigating the use of the fpnew with the CV32E40P.  I believe they have made some progress, and if you like I can make an introduction.

I hope this helps.   Let us know if you would like more information.

closed 2021-04-27 by MikeOpenHWGroup

diggi0330, 2023-12-21:
Hi @MikeOpenHWGroup, could you please point me to any resources to integrate fpnew (cvfpu) with cv32e40p to work with (or without) verilator? 

Please note that for my design (as detailed above as 1), I successfully verified FP operations with FPU = 1 and ZFINX = 1. But, since these operations essentially use base core without FPU support, the instruction cycle count is huge. Hence, I require cvfpu support.

pascalgouedo, 2023-12-22:
In CV32E40Pv2, there is a new top which includes both the core and fpu: cv32e40p_top
It is fully verified (even fpu instructions).
A new test (matmul_32b_float) is an example which compiles and executes C program with FPU support. It needs to be used with pulp_fpu configuration:
makeuvmt test CFG=pulp_fpu TEST=matmul_32b_float USE_ISS=NO

MikeOpenHWGroup, 2023-12-22:
Also, please note that my original statement from 2021 is no longer true:
> The current version of the CV32E40P has not been verified with the floating point unit.

@pascalgouedo and the team at Dolphin Design are working on v2.0.0 which does include support for the FPU.  Have a look at the lastest User Manual ([link](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/)).

diggi0330, 2023-12-24:
Thank you @pascalgouedo @MikeOpenHWGroup for your quick comment.

> In CV32E40Pv2, there is a new top which includes both the core and cvfpu: cv32e40p_top It is fully verified (even fpu instructions). A new test (matmul_32b_float) is an example which compiles and executes C program with FPU support. It needs to be used with pulp_fpu configuration: makeuvmt test CFG=pulp_fpu TEST=matmul_32b_float USE_ISS=NO

@pascalgouedo could you please help me locate the CV32E40Pv2 GitHub repository? I can't find it online. This will allow me to test the FPU support.

Thanks again.