openhwfoundation/cv32e40p #640: Removing the PMP file from the RTL folder

davideschiavone, 2021-05-03:
shall we remove the PMP file from the RTL folder?

if people look for it for older version of the core, we can still point to an old commit where the file is present.

The current RTL fails to be formatted by verible and it containts casex from systemverilog, which makes the life of some EDA tools hard.

In addition, as today, there is no plan to integrate the PMP into CV32E40P.

best
Davide

Silabs-ArjanB, 2021-05-03:
I would say 'yes', let's remove that file and any references to it.

MikeOpenHWGroup, 2021-05-03:
Its my understanding that the reason this question came up is that `cv32e40p_pmp.sv` contains a `casex` statement.  As far as I am aware, it is the only occurrence of a casex statement in any CORE-V RTL.

Just FYI, it is my understanding that the RTL coding style guidelines for CORE-V IP is essentially the lowRISC Ibex SystemVerilog coding style guide.  Under [wildcards-in-case-items](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#wildcards-in-case-items), these guidelines state that casex statements should not be used.

davideschiavone, 2021-05-03:
Not only that. Also the verible-format tool does not pass (the casex is also present in the tracer and verible was able to pass thought it).
In addition, none is using that file anyway. It may create confusion to have a SV file in the RTL folder which is not used or instantiated anyway.

davideschiavone, 2021-05-20:
close by #645