openhwfoundation/cv32e40p #642: Issue with the constraints file

Mar486, 2021-05-07:
Hi,

I have a problem regarding the constraints file. 
DC compiler gives me warnings regarding these two lines in the constraint file:

set all_other_input_ports  [remove_from_collection [all_inputs]  [get_ports [list $all_clock_ports $obi_input_ports $irq_input_ports $early_input_ports]]]
set all_other_output_ports [remove_from_collection [all_outputs] [get_ports [list $all_clock_ports $obi_output_ports $sleep_output_ports $irq_output_ports]]]

the warnings:

Warning: Can't find ports matching 'instr_gnt_i instr_rvalid_i instr_rdata_i* data_gnt_i data_rvalid_i data_rdata_i*' in design 'cv32e40p_core'. (UID-95)
Warning: Can't find ports matching 'debug_req_i boot_addr_i* mtvec_addr_i* dm_halt_addr_i* hart_id_i* dm_exception_addr_i*' in design 'cv32e40p_core'. (UID-95)
Warning: Can't find ports matching 'instr_req_o instr_addr_o* data_req_o data_we_o data_be_o* data_addr_o* data_wdata_o*' in design 'cv32e40p_core'. (UID-95)
Warning: Can't find ports matching 'irq_ack_o irq_id_o*' in design 'cv32e40p_core'. (UID-95)

Mar486, 2021-05-09:
1. reading the libraries.
2. analyze the RTL files ordered according to the manifest file.
3. elaborate.
4. current_design
5. reading the constraints file (sdc file).
6. link
7. compile_ultra

zorrolee777, 2021-05-19:
Hi,

Has this been fixed? I got similar issues.

zorrolee777, 2021-05-19:
And I also saw there's a draft version DC script here: https://github.com/openhwgroup/cv32e40p/pull/305
Is there any complete version? Thanks!

Silabs-ArjanB, 2021-05-19:
@zorrolee777 We do not have the intention to provide synthesis scripts for specific vendors, just example constraints. (I don't know if the stated issue has been fixed; I do not have access to the stated tool.)

zorrolee777, 2021-05-20:
> > hello @Mar486 , can you please provide what are all the commands you type in DC?
> 
> 1. reading the libraries.
> 2. analyze the RTL files ordered according to the manifest file.
> 3. elaborate.
> 4. current_design
> 5. reading the constraints file (sdc file).
> 6. link
> 7. compile_ultra

Hello @Mar486. Any luck to fix this?

Mar486, 2021-05-21:
> Hi,
> 
> Has this been fixed? I got similar issues.

I replaced these two lines by the following lines:

set f [all_inputs]
set a [get_ports $all_clock_ports]
set b [get_ports $obi_input_ports]
set c [get_ports $irq_input_ports]
set d [get_ports $early_input_ports]

set a_b [add_to_coll $a $b]
set abc [add_to_coll $a_b $c]
set abcd [add_to_coll $abc $d]

set all_other_input_ports [remove_from_coll $f $abcd] 

set q [all_outputs]
set m [get_ports $all_clock_ports]
set n [get_ports $obi_output_ports]
set o [get_ports $sleep_output_ports]
set p [get_ports $irq_output_ports]

set m_n [add_to_coll $m $n]
set mno [add_to_coll $m_n $o]
set mnop [add_to_coll $mno $p]

set all_other_output_ports [remove_from_coll $q $mnop]

davideschiavone, 2021-05-21:
can you please provide more details about the points 1 to 6?  I would need to replicate your commands to understand the issue

MikeOpenHWGroup, 2021-05-21:
@Mar486, would it be possible for you to share your script?

Mar486, 2021-05-21:
> can you please provide more details about the points 1 to 6? I would need to replicate your commands to understand the issue

> @Mar486, would it be possible for you to share your script?

set design cv32e40p_core

set_app_var search_path "/home/standard_cell_libraries/NangateOpenCellLibrary_PDKv1_3_v2010_12/lib/Front_End/Liberty/NLDM"

set_app_var link_library "* NangateOpenCellLibrary_ss0p95vn40c.db"
set_app_var target_library "NangateOpenCellLibrary_ss0p95vn40c.db"

sh rm -rf work
sh mkdir -p work
define_design_lib work -path ./work
set hdlin_sverilog_std                  2009

    analyze -library work -format sverilog ../rtl/cv32e40p_register_file_latch.sv
    analyze -library work -format sverilog ../rtl/include/cv32e40p_apu_core_pkg.sv
    analyze -library work -format sverilog ../rtl/include/cv32e40p_pkg.sv
    analyze -library work -format sverilog ../rtl/include/cv32e40p_fpu_pkg.sv
    analyze -library work -format sverilog ../bhv/include/cv32e40p_tracer_pkg.sv
    
    analyze -library work -format sverilog ../rtl/cv32e40p_alu.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_alu_div.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_ff_one.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_popcnt.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_compressed_decoder.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_controller.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_cs_registers.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_decoder.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_int_controller.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_ex_stage.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_hwloop_regs.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_id_stage.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_if_stage.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_load_store_unit.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_mult.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_prefetch_buffer.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_prefetch_controller.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_obi_interface.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_aligner.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_sleep_unit.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_core.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_apu_disp.sv
    analyze -library work -format sverilog ../rtl/cv32e40p_fifo.sv
   
    analyze -library work -format sverilog ../bhv/cv32e40p_sim_clock_gate.sv
    analyze -library work -format sverilog ../bhv/cv32e40p_wrapper.sv
    
    
elaborate $design -lib work
current_design 

check_design

source ./cons/cv32e40p_core.sdc
link

compile_ultra -timing_high_effort_script