openhwfoundation/cv32e40p #647: Master branch not working with Verilator

aignacio, 2021-06-03:
Hi everyone,
I'm using the master branch to build a SoC with CV32E40P core but I'm seeing an odd issue with Verilator, I opened a ticket on Verilator to check for support but I'm not sure they can understand the semantic error. Is there a branch with Verilator running fine?

[Link for the issue on Verilator](https://github.com/verilator/verilator/issues/2998)

MikeOpenHWGroup, 2021-06-03:
Hi @aignacio, thanks for the issue.  Can you provide me with the _exact_ command-lines you are using?   Also, what version of Verilator are you using?

MikeOpenHWGroup, 2021-06-04:
Hi again @aignacio.  If you run the CV32E40P from the CORE-V verification environment (https://github.com/openhwgroup/core-v-verif), it compiles and runs fine with Verilator v4.202:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32e40p/sim/core
$ make
```
This will clone a specific tag of CV32E40P from its repo (the latest, as of today), built and run "hello-world" using Verilator.  The READMEs will give you more information for running other tests.

aignacio, 2021-06-05:
Hi @MikeOpenHWGroup,

I just call Verilator through a makefile passing CV32 rtls as argument and observed this:
```bash
- V3Ast.cpp:1138:     Dumping output_verilator/Vsimple_tile_002_cells.tree
%Error: Internal Error: rtl/cv32e40p/rtl/cv32e40p_id_stage.sv:31:22: ../V3Broken.cpp:253: Broken link in node (or something without maybePointedTo): !m_packagep || !m_packagep->brokeExists()
-node: PACKAGEIMPORT 0x7f9a09d825c0 <e95652> {x31aw}  * -> PACKAGE 0x7f9a19eb2270 <e314138> {e26aj}  cv32e40p_pkg  L7 [LIB] [NONE]
   31 |   import cv32e40p_pkg::*;
      |                      ^~
                        ... See the manual and https://verilator.org for more assistance.
- V3Ast.cpp:1138:     Dumping output_verilator/Vsimple_tile_990_final.tree
- V3Broken.cpp:318:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:226:statsReport:
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
make: *** [output_verilator/Vsimple_tile.mk] Error 255
```
[Repository](https://github.com/aignacio/mpsoc_example)
Verilator version: Verilator 4.106 2020-12-02 rev v4.106

aignacio, 2021-06-05:
I'll try to use the same approach as you said in the core-v-verif repo, let's see how it goes..

MikeOpenHWGroup, 2021-06-05:
Your Makefile compiles everything it finds with an "sv" extension.  That will not work with Verilator as the behavioral tracer and wrapper use some SV constructs that Verilator does not support.  A better idea is to use the manifest file at the top of the cv32e40p repo.

davideschiavone, 2021-06-28:
@aignacio - can I close this issue?

aignacio, 2021-06-28:
Hi, couldn't have time to check but if I got in the same issue I'll re-open it. Tks anyway

closed 2021-06-28 by aignacio