openhwfoundation/cv32e40p #649: JALR instruction, jump error

apollojason, 2021-06-24:
Dear Guys：
         These days i used the cv32e40p core, with FPU.
          When i use the Core，there is a JALR instruction,   the jump result is wrong.
          I checked the waveform, the main reason i thought is  data dependency.
          The JALR depends on the result of an immediately preceding instruction.
          plesase see my trace log.
         
[image]
PC 48002582  fmv.s.x,  the fmv.s.x is a multicycle instruction.
when the fmv.s.x is in WB stage, the jalr (4800258c) is in decode stage.
When the jalr is in the decode stage, the decoder will judge the current PC instruction is a jump instrtuion，the signal ‘branch’ will be set.
Why？ The fmv.s.x used the regfile ID is 0x2a, but the jalr used the 0x1. So, the signal ‘’reg_d_alu_is_reg_a_id‘’ in id_stage.sv will be invalid.
[image]

 So, the ‘jr_stall’ logic can't stall the pipeline. The wrong ‘branch’ will cause the wrong jump PC.
[image]

Have you ever considered the auipc and jalr instrution combined wiht multicycle instruction?
I think if the multicycle insrtucion used the different regfile, it will influence the jalr instrution, maybe.

Could you help check the waveform? 
[image]
                                      

                                                                                                                                                                                   Jason Wu

davideschiavone, 2021-06-24:
Hello @apollojason 

thanks for the detailed report - FYI, we did not verify the FPU, in addition, we are about to change the communication channel (APU) that the core uses to talk with the FPU with a new one defined here: https://github.com/openhwgroup/core-v-xif/blob/main/doc/index.md

Still, I would like to have a closer look at your problem, can you please share a VCD file, the OBJDUMP and the Binary file so that I can be more helpful? Are you running your application on the example_tb/core folder? or where?

thanks again for giving such a detailed description

apollojason, 2021-06-25:
Dear @davideschiavone:
       Thank you very much!
       I know that you have move the FPU outside the CPU core, and communicate by the APU interface.
       But i don't think this change can avoid the JALR stall.
       As i understand it, the key factor is the jr stall control strategy, especially the multicycle instruction will change the regfile ID.
       I‘m curious to know whether the  integer Multiplication、Division will cause this problem?

       The cv32e40p_jalr_jumppc_err.vcd.vpd is the waveform,     
       You can use the cmd 'dve -vpd ' open the wave.
[cv32e40p_jalr_jump_pc_err.vcd.zip](https://github.com/openhwgroup/cv32e40p/files/6713022/cv32e40p_jalr_jump_pc_err.vcd.zip)
     
       thanks again
                                                                                                                       Jason Wu

apollojason, 2021-06-28:
Dear @davideschiavone ：
       Have you ever checked the waveform?
       I hope you can give me some suggestions.
                                        Jason Wu

davideschiavone, 2021-06-28:
Hello,

not yet - I have been quite busy lately, I will try to have a look at it this week.

Meanwhile we are proceeding with projects to verify CV32E40P+FPU. Thus, this bug is definitively useful to address asap.

I will let you know

Davide

davideschiavone, 2021-06-29:
Can you please provide a normal VCD? I would like it to open it with GTKwave or Questa for example - I can't with that file

davideschiavone, 2021-06-29:
I am using the FPNEW IP that CV32E40P clones, the fmv instruction takes 1 cycle in my case (and it seems also in your case as the CYCLE number in the tracer increments only by one)

`
   3835ns             379 0000037c f0060553 fmv.s.x          f10, x12            f10=00000000 x12:00000000
   3845ns             380 00000380 00305097 auipc            x1, 0x305000         x1=00305380
   3865ns             382 00000384 bc4080e7 jalr             x1, x1, -1084        x1=00000388  x1:00305380
`

Here the jalr instruction uses as X1 the right value and it runs 1 cycle later as the first cycle it sees a STALL from the AUIPC.

I see exaclty the same situation in your tracer, so I do not understand what the problem is in your case. 

In my case, the address is 00305380 + FFFF FBC4‬ (whic is -1084), and it is. 

In your case it should be 0x4830714C‬ - which is not .

Can you tell me whether you are using the master branch or which commit, if you are using the FPNEW, etc

davideschiavone, 2021-07-09:
do you have any update on this @apollojason ?

davideschiavone, 2021-07-20:
@apollojason ,as I haven't received any feedback, I will close this issue - feel free to re-open it if your problem hasn't been fixed.
Thanks a lot, 
Davide

closed 2021-07-20 by davideschiavone