openhwfoundation/cv32e40p #651: Recommended Compiler + Flags for Xcorev/pulp extensions

RTLhamster, 2021-07-07:
Hi there, 

I'm trying to get the cv32e40p running with the xcorev/pulp extensions, 
but it seems that the CORE-V GCC is not using these extensions. 

In simulation/verification setups I didn't find an example actually using the *_xcorev march flag. 
The cv32e40p user manual just tells:
> To use such instructions, you need to compile your SW with the CORE-V GCC compiler.

Can you please comment on the current compiler support and recommended compiler settings.

I've also opened a issue in corev-gcc repro [corev-gcc ISSUE #1](https://github.com/openhwgroup/corev-gcc/issues/1), 
but it doesn't seem actively maintained (last commit 14 Sep 2020).

Thank you for your attention

RTLhamster, 2021-07-08:
As mentioned by @jeremybennett in [corev-gcc ISSUE #1](https://github.com/openhwgroup/corev-gcc/issues/1)
there is currently no support for automatic insertion of xcorev extension by the compiler. 

Unlike the [user manual](https://cv32e40p.readthedocs.io/en/latest/corev_hw_loop/#hwloop-specs) is telling us:

>The CORE-V GCC compiler uses HWLoop automatically without the need of assembly. The mainline GCC does not generate any CORE-V instructions as for the other custom extensions. 

In my opinion this should be corrected or clarified.

MikeOpenHWGroup, 2021-07-08:
Hi @jeremybennett, can you comment?

bluewww, 2021-08-04:
@RTLhamster Until the official CORE-V GCC catches up, you could try PULP GCC (v2.4.0 based on gcc 9.2.0) here https://github.com/bluewww/riscv-gnu-toolchain/. We use it at ETH to compile for the cv32e40p. It supports `*_xcorev`, but you probably need to disable hardware loops (`-mno-pulp-hwloop`). Here is an overview https://github.com/bluewww/riscv-gnu-toolchain/blob/master/pulp.md

pascalgouedo, 2024-03-25:
Hi,
Using Embecosm 2024 toolchains, just use add the following to your march option:
"_xcvhwlp_xcvmem_xcvmac_xcvbi_xcvalu_xcvsimd_xcvbitmanip"

e.g. rv32imc_zicsr_zifencei_xcvhwlp_xcvmem_xcvmac_xcvbi_xcvalu_xcvsimd_xcvbitmanip

closed 2024-03-25 by pascalgouedo