openhwfoundation/cv32e40p #652: Wrong multiplication instructions' result stored in X when a preceding Zfinx instruction is executed

shetalani, 2021-07-09:
### Issue Description
M extension `MULHU` updates the integer register file with a wrong value when `FPU` and `PULP_ZFINX` are set. Same is true for `MULHU` and `MULHSU`. 

### Component
**Component:RTL**

### RISC-V Specification

> 24: (Zfinx spec) The Zfinx extension adds all of the instructions that the F extension adds, except for the transfer
instructions FLW, FSW, FMV.W.X, FMV.X.W, C.FLW[SP], and C.FSW[SP].

> 24: (Zfinx spec) The Zfinx variants of these F-extension instructions have the same semantics, except that whenever
such an instruction would have accessed an f register, it instead accesses the x register with the
same number.

> 11.3: Except when otherwise stated, if the result of a floating-point operation is NaN, it is the canonical NaN. The canonical NaN has a positive sign and all significand bits clear except the MSB, a.k.a. the quiet bit. For single-precision floating-point, this corresponds to the pattern 0x7fc00000.

### Steps to Reproduce

As shown below, the following sequence of instructions happens: 
> fdiv.s f8, f0, f0, RMM -> mulhu x15, x28, x29 

The first instruction updates the `X8` register (`Zfinx` is set) with the value `32'h7fc00000` (`NaN` result of division by 0). However, this interferes with the execution of the next `MULHU` instruction decoded `@t` and thus a wrong value of `32'h7fc0`, instead of `32'h0`, is stored in `X15` when the multiplication unit is ready `@t_mulh` (5 clock cycles execution). 

[image]

[mulhu.vcd](https://github.com/openhwgroup/cv32e40p/files/6791508/wrapper_i.core_i.RV_chk.ops.RV32M.zip)

**Top Level Parameters**
```
cv32e40p_tb_subsystem #(
    .INSTR_RDATA_WIDTH(32),
    .RAM_ADDR_WIDTH   (20),
    .BOOT_ADDR        ('h180),
    .PULP_XPULP       (0),
    .PULP_CLUSTER     (0),
    .FPU              (1),
    .PULP_ZFINX       (1),
    .NUM_MHPMCOUNTERS (1),
    .DM_HALTADDRESS   (32'h1A110800)
)
```
The wrapper of [example_tb/core](https://github.com/openhwgroup/cv32e40p/tree/master/example_tb/core) is used to instantiate the core with fpnew. https://github.com/openhwgroup/cv32e40p/commit/523b80f0635da833417c8b2ff0f0afe39cda2aaa

---
Product: OneSpin 360 DV-Verify
App: Processor Verification App
Tool's version: 2021.1.2

davideschiavone, 2021-07-09:
Hello,

thanks a lot for the report. Just two notes:

1) The RVF and PULP_Zfinx instructions set haven't been verified yet as stated in the documentation - so thanks a lot for the test as we will verify them soon, and thus this bug should be fixed

2) The PULP_Zfinx is a custom extension - it was born before the Zfinx - but they are not exactly the same - so the SPECs (which are not written yet as part of an upcoming project) may be different 

I flag this as bug as get back to you once proceeding with the fix

Thanks 
Davide

pascalgouedo, 2023-10-09:
Resolved with PR #860

closed 2023-10-09 by pascalgouedo