openhwfoundation/cv32e40p #653: OBI be signal not part of instruction bus

jm4rtin, 2021-07-22:
In the OBI 1.2 specification it describes the `be` signal as being used for both reads and writes. It makes no exception for a bus that may only read (and only read 32-bit words). It does not mark the signal as optional.

That means either the CV32E40P instruction bus should include the `be` signal as `instr_be` (tied to 4'b1111) to comply with the OBI specification or the OBI specification should be updated to make `be` optional with the default/inferred value of all ones (4'b1111 for a 32-bit bus).

Silabs-ArjanB, 2021-07-22:
Hi @jm4rtin , table 6 mentions the default 4'b1111 tie off for the 'be' signal.

jm4rtin, 2021-07-22:
Sorry, I missed that section of the OBI specification.

closed 2021-07-22 by jm4rtin