openhwfoundation/cv32e40p #668: Add FPU integration information to the User Manual

MikeOpenHWGroup, 2021-09-23:
The [Floating Point Unit chapter of the User Manual](https://github.com/openhwgroup/cv32e40p/blob/master/docs/source/fpu.rst) contains very little information about how the CV32E40P and FPU should be configured to ensure proper operation.  There are parameters in both the CV32E40P (see [cv32e40p_pkg.sv](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/include/cv32e40p_pkg.sv#L726)) and FPU (see [fpnew_pkg.sv](https://github.com/pulp-platform/fpnew/blob/develop/src/fpnew_pkg.sv#L256)) that must be set to "useful" values.

The updates to the Floating Point Unit chapter should include:
- Definition of the relevant parameters in the both the CV32E40P and FPnew.
- A description of which parameter sets have been used to verify the operation of the CV32E40P+FPU, similar to the existing [Parameters](https://github.com/openhwgroup/cv32e40p/blob/master/docs/source/integration.rst#parameters) section of the Core Integration chapter of the User Manual.

davideschiavone, 2021-09-24:
@pascalgouedo can probably help with his specific use-case.

In the testbench, for what concerns the single-core, here are the parameters used: https://github.com/openhwgroup/cv32e40p/blob/master/example_tb/core/cv32e40p_fp_wrapper.sv#L62

MikeOpenHWGroup, 2021-09-24:
Thanks @davideschiavone.  We are aware of the parameters used in the single-core testbench.  The purpose of this issue is to have these recorded and documented in the User Manual to the same level of detail as we currently have captured for the other parameters in CV32E40P.  This is important for two reasons:
1. The verification team needs to have a single-point of reference for the proper configuration of the core and the FP unit.  (An example testbench found on the repo is not sufficient for that.)
2. Users of the core need to understand what has (and has not) been verified.

pascalgouedo, 2021-09-28:
Hi,
Compared to the single-core test-bench parameter values, I just changed:
- EnableNaNBox to 1 even if I guess it is not necessary. But it allows to have all FPnew configs to be aligned on this point.
- FP operation latencies with more values because I think keeping only 0 (and 1 for DIV/SQRT) will maybe reduce overall max frequency. We are going to make different experiments of PPA analysis with respect to technologies and frequencies to understand the impact of FPU on that point. By the way I just corrected CV32E40P + FPnew DIVSQRT latency in the shared document because I forgot to make it coherent with other latencies.

pascalgouedo, 2023-08-04:
Resolved with PR #791 and available since cv32e40p_v1.3.0 User Manual

closed 2023-08-04 by pascalgouedo