openhwfoundation/cv32e40p #677: How to simulate the cv32e40p_tracer with verilator ?

mohamedElbouazzati, 2021-12-21:
Hello @openhwgroup, 

i'm using the core-v-verif to simulate the cv32e40p with verilator, i found the tracer block in cv32e40p documentation which can provide us a log file of  traces which something that interests me.

As provided in documentation i added this line` CV32E40P_TRACE_EXECUTION` in the `tb_top.sv` located in `/core-verif/cv32e40p/tb/core ` i'm a beginner with SystemVerilogi,  Is this line enough to get the log file?

I tried with `make` in `/core-v-verif/cv32e40p/sim/core/` i did not get the log file ?

Also in `cv32e40p_manifest.flist` i added the path of  two block of tracer  in` /cv32e40p/bhv/`

Any help please
Thanks

MikeOpenHWGroup, 2021-12-21:
Hi @mohamedElbouazzati.  The best place to look is the manifests and Makefiles for the UVM environment.  The core testbench is a demonstration vehicle which does not have either a tracer or reference model.  Note that the UVM environment will not compile using Verilator, so you will require a commerical SV simulator.

Having said that, it should be possible to add the Tracer to the core testbench and run it with Verilator.  In addition to adding the correct files to the manifest (which you appear to have done), you will also need to set the `CV32E40P_TRACE_EXECUTION` macro, either with a "tick define" in your testbench or on the command-line.  Probably the best place to do that is to add it to the [Verilator configuration in the Makefile](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/sim/core/Makefile#L150).

mohamedElbouazzati, 2021-12-28:
Hello @MikeOpenHWGroup , 

As you said i tried to add` 'define CV32E40P_TRACE_EXECUTION` in the testbench for verilator but still have many errors, i'm trying to use another version of verilator and will post the error maybe here or in another issue.

Thanks

bluewww, 2021-12-28:
I don't think the version of verilator matters too much. Verilator doesn't support the SystemVerilog features used in the tracer code.

MikeOpenHWGroup, 2021-12-28:
> Having said that, it should be possible to add the Tracer to the core testbench and run it with Verilator.

(embarrassed cough)  Ah, my apologies @mohamedElbouazzati.  @bluewww is correct.  The CV32E40P tracer uses SV class data members and several UVM macros.  Verilator does not support these code constructs.   Sorry for providing you with incorrect information.  You might want to try Xilinx Vivado.  The free version of their SV simulator supports _almost_ all of IEEE1800-2017 and is known to support UVM environments.  (Note that I have not tried this.)

mohamedElbouazzati, 2021-12-29:
Hi, 

Thank you so much @MikeOpenHWGroup and @bluewww for this clarification,for your information : i found also on github  in `antmicro/verilator-dynamic-scheduler-examples` a verilator version that support now some of UVM macros (randomize, fork, events, ..) and it is under development to support full UVM environment, i will try this also with the tracer and give you a feedback, maybe it will be useful for community using the core-v-verif repo with verilator.

Thank you :+1:

closed 2021-12-29 by MikeOpenHWGroup

mohamedElbouazzati, 2022-01-11:
Hi @MikeOpenHWGroup, 

Finally, I got good results with the Xilinx Vivado simulator. 

Thank you

MikeOpenHWGroup, 2022-01-11:
That is terrific @mohamedElbouazzati!  I am impressed by your tenacity.

It would be a valuable contribution to OpenHW if you could provide the Makefile that works with Vivado.