openhwfoundation/cv32e40p #678: Illegal Instruction Error while Running a FFT application?

LatifAkcayGithub, 2021-12-30:
Hello,

We are trying to run 2048 point FFT operation on Pulpissimo (CV32E40P).
While lower number of FFT codes run successfully ( 128, 256, 1024), 2048 points version code is stuck.
We think that it may be a memory problem. So, we extend the instruction memory(PRI_1) to 128 K and Data memory (PRI_0) to 256 K.
After the memory extension, the lower FFT points again run successfully. However, 2048 points version is again stuck.
We also extend the stack size defined in link.ld from 0x800 to 0x4000. But, the code is again stuck.
In modelsim simulation, when running FFT operations, an illegal fetch adress (an unalligned adress) is generated and then the core could not find an instruction word at that adress. Then, the core starts gving "illegal instruction" error.
So, what could be the reason behind this issue? And how can we solve it?
(We are using KissFFT C library for our fft applications)

Thank you.

bluewww, 2021-12-30:
In modelsim you can check what the address of the illegal instruction is that should tell you already a lot of what is going wrong. Also this is not a cv32e40p problem as far as I can tell so this issue should be closed.

davideschiavone, 2022-01-05:
Agree with @bluewww - did you only change the linker script or did you also change the HW @LatifAkcayGithub ?

davideschiavone, 2022-12-23:
Closing as not activity has been shown since almost 1 year - feel free to reopen if needed

closed 2022-12-23 by davideschiavone