openhwfoundation/cv32e40p #684: reg_fp* values are wrong when PULP_Zfinx is enabled

abukharmeh, 2022-03-10:
https://github.com/openhwgroup/cv32e40p/blob/b7c91867b2f611013dbe98150e745b1a2ff6b7c6/rtl/cv32e40p_decoder.sv#L1412-L1414

The values of reg_fp_* are assigned incorrectly when Zfinx is enabled, and then overridden in Register file, and now in #683  in the tracer. 

I think PULP_ZFINX parameter value should be passed to the decoder module, and these values should be assigned to `!PULP_ZFINX`

pascalgouedo, 2022-03-14:
I agree that the FP RF use or not should be made as soon as possible in RTL logic to avoid to duplicate things later on in ID and in tracer.
In fact moving/merging fregfile_ena from ID to Decoder.

abukharmeh, 2022-03-14:
This looks really good, thank you very much. 

I wonder if we could do direct assignment from parameter to logic, i.e reg_fp_a_o  =  1'(!ZFINX_EN)

I had a quick look at https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md and could not find anything that discourage that ?

commit 2fe771be (2022-04-20): Fix #684
commit f684a361 (2022-04-20): Merge pull request #685 from pascalgouedo/dd_pgo_issue_684

Fix #684