openhwfoundation/cv32e40p #689: Building Project and Verification

mattvlsi, 2022-04-03:
I'm a newby on Risc-v cores and its verification. I cloned core-v-verif and cv32e40p repos and run make at 
**~/core-v-verif/cv32e40p/sim/core**:
I'm getting those errors:
**~/core-v-verif/mk/Common.mk:83: Must define a DPI_DASM_SPIKE_REPO to use the common makefile
~/core-v-verif/mk/Common.mk:253: *** CV_SW_TOOLCHAIN not defined in either the shell environment, test.yaml or cfg.yaml.  Stop.**

However I downloaded COREV toolchain on /opt/corev and defined an environment path on .bashrc as _export COREV=/opt/corev_ 

Also I want to build a project of the core in Vivado. Do I need to do it manually or is there any ready-to-use script for building RTL project in Vivado or ModelSim?

Thanks.

MikeOpenHWGroup, 2022-04-03:
Thanks for your issue @mattvlsi.  Have you had a look at the [Quick Start Guide](https://core-v-docs-verif-strat.readthedocs.io/en/latest/quick_start.html) in the Verification Strategy document on ReadTheDocs?  Following the QSG _should_ get you running (please let me know if it does not).

Defining **DPI_DASM_SPIKE_REPO** is only required if you are using the ISACOV functional coverage model, which the core testbench cannot use, so you are good there.  (We really should make that more clear.)

**CV_SW_TOOLCHAIN** is one of four [required shell env variables](https://github.com/openhwgroup/core-v-verif/tree/master/mk#required-corev-environment-variables) you will need to set to run core-v-verif.  That link will tell you what you need.

We do not yet have Vivado running on core-v-verif.  It would be a great contribution to core-v-verif if you could provide a set of `make` targets for Vivado.   Let me know if you are interested in pursuing that.

mattvlsi, 2022-04-05:
Thank you @MikeOpenHWGroup for clear explanation. If I can build on Vivado and run testbenches on it, definitely I will let you know. I have one more question. I need to have Double Precision FPU on my Riscv. I saw Double Precision option in fpnew source files. Is it adaptable on cv32e40p or is there any other open source core that implements Double FP (RV32IMFD)? Thanks.

MikeOpenHWGroup, 2022-04-05:
The short answer to your question is "yes, but the F and D extensions are not yet fully verified".  If you are interested in more details please reach out to me at my email address, or better yet on OpenHW's [MatterMost](https://mattermost.openhwgroup.org/) chat service.

If this issue is resolved please mark it Closed, and I'll look forward to your pull-request to bring in the Vivado simulator.

closed 2022-04-05 by mattvlsi