openhwfoundation/cv32e40p #701: Question about RTL bloc : cv32e40p_alu_div.sv file

weik56100, 2022-04-27:
I try to understand which algorithm is implemented in the calculation of the division on signed integers. (cv32e40p_au_div.sv)

As the division is not constant time (3-35 cycles), I would like to modify the algorithm to make it constant time.

Two ways for me:

Worst case:

- Take the worst case and make the division always take 35 cycles under any circumstances.

Best case:

- By a much shorter constant time calculation, but still the same number of cycles.

If you have information about the implementation that is used in the RTL block, I would be interested.

Thanks

weik56100, 2022-05-11:
Update:

I was able to set up the first case, everything works correctly after my various tests.

For the second case I would like to implement a non-restoring division in base 4. 

From which source/algorithm was the RTL block of the division built?

I am trying to understand what is in place so I can modify it later.

Thanks

MikeOpenHWGroup, 2022-05-11:
Hi @weik56100.  I did not design the divider in CV32E40P, so I do not have a definitive answer for you, but I believe the algorithm used is "non restorative long division".  [Here](https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9340245) is a paper that summarizes the characteristics of multiple division algorithms.

closed 2022-08-08 by davideschiavone