openhwfoundation/cv32e40p #709: Counting Leading bits: Zeros or Ones or Max?

a7med7asan15, 2022-06-07:
Hello Everyone! 
My question is regarding the instruction: cv.clb rD,rs1 It's an XPULP Instruction

Instruction Description: rD = count leading bits of rs1 Note: This is the number of consecutive 1’s or 0’s from MSB. Note: If rs1 is 0, rD will be 0.

It's not clear what it counts. Does it count the maximum leading bits whether zeros or ones?

pascalgouedo, 2022-06-07:
Hi Hassan,

Starting from the MSB of the operand, it counts all the bits with the same polarity until it finds a bit with the opposite polarity.
Examples:

00000100010010... = 5
11111111010010... = 8

Pascal.

From: Ahmed Hassan ***@***.***>
Sent: mardi 7 juin 2022 12:21
To: openhwgroup/cv32e40p ***@***.***>
Cc: Subscribed ***@***.***>
Subject: [openhwgroup/cv32e40p] Counting Leading bits: Zeros or Ones or Max? (Issue #709)

Hello Everyone!
My question is regarding the instruction: cv.clb rD,rs1 It's an XPULP Instruction

Instruction Description: rD = count leading bits of rs1 Note: This is the number of consecutive 1's or 0's from MSB. Note: If rs1 is 0, rD will be 0.

It's not clear what it counts. Does it count the maximum leading bits whether zeros or ones?

a7med7asan15, 2022-06-07:
It's clear now. 
Thanks Pascal

closed 2022-06-07 by a7med7asan15