openhwfoundation/cv32e40p #718: Maybe an error: number of inter registers is only 16

Xuequan, 2022-08-08:
I am a newer.  Please forgive me if i am wrong. BTW,thanks for great project.
1. file : https://github.com/pulp-platform/cv32e40p/blob/master/rtl/cv32e40p_register_file_ff.sv
2. Command line : line 66 & 68

pascalgouedo, 2022-08-08:
Hi,

CV32E40P is RV32I so its register file has 32 integer registers.
If FPU is enabled then an additional register file is present with 32 floating-point registers.

Regards.

closed 2022-08-08 by davideschiavone

Xuequan, 2022-08-12:
Dear Davide Schiavone,

Thanks for your quick reply. But still confused.  Yes, register number is 32. 

Code 31 & 66 as below: 

 parameter ADDR_WIDTH = 5,

  localparam NUM_WORDS = 2 ** (ADDR_WIDTH - 1);

So, NUM_WORDS = 2^4 = 16.  Not 32 ?  

This is where I don't understand. 

Closed #718 as completed.

pascalgouedo, 2022-08-12:
Hi,

Value 5 for parameter ADDR_WIDTH is a default value if this parameter is not set when cv32e40p_register_file block is instantiated at upper level.
But this parameter default value is overridden to 6 in cv32e40p_id_stage when instantiating cv32e40p_register_file.

Regards,
Pascal.

Xuequan, 2022-08-12:
Dear Davide Schiavone,

Got it! 
Thanks for your quick reply!  It is very nice of you.
I hope all is well with you.