openhwfoundation/cv32e40p #730: Custom Xpulp instructions update register file wrongly | Forwarding issue

salaheddinhetalani, 2022-09-27:
### Issue Description
Custom XPULP instructions, CV.ADDxNR and CV.SUBxNR, read invalid source operand in case they are preceded by a multicycle F instruction and their source register is the same destination register of the floating point one, thus resulting wrong register file update.

### Component
**Component:RTL**

### RISC-V Specification
> The Zfinx extension adds all of the instructions that the F extension adds, except for the transfer instructions FLW, FSW, FMV.W.X, FMV.X.W, C.FLW[SP], and C.FSW[SP].

>The Zfinx variants of these F-extension instructions have the same semantics, except that whenever such an instruction would have accessed an f register, it instead accesses the x register with the same number.

### Steps to Reproduce

As shown below, the following sequence of instructions happens: 
> fdiv.s x5, x21, x4, RTZ -> cv.addunr x13, x1, x5 

The instruction `cv.addunr` is decoded at `t##0` and executed updating the integer register file at `t##1` while setting the wrong result of `32'h189` into `x13`. This is due to reading the old value `32'h20e57` of `x5` instead of the new one of `32'h0` updated at `t##2` by the previously fetched `fdiv.s`. 

[image]

**Top Level Parameters**

```
cv32e40p_wrapper #(
    .PULP_XPULP       (1),
    .PULP_CLUSTER     (1),
    .FPU              (1),
    .PULP_ZFINX       (1),
    .NUM_MHPMCOUNTERS (1)
)
```
Git Hash: TBU
Flist: [cv32e40p_fpu_manifest.flist](https://github.com/openhwgroup/cv32e40p/blob/master/cv32e40p_fpu_manifest.flist.) 
VCD: [bug_26.vcd](https://github.com/openhwgroup/cv32e40p/files/9657041/bug_26.zip)

---
Product: OneSpin 360 DV-Verify
App: Processor Verification App
Tool's version: 2022.3_1

davideschiavone, 2023-08-03:
can we close this as this one is now merged? 6cfb7438ba62fb433823c85d04e2e001d56e5b81

pascalgouedo, 2023-08-03:
Work in on-going on bug resolution w/ formal verification guy.
We will close all resolved opened issue together once confirmed by formal runs.

pascalgouedo, 2023-10-09:
Resolved with PR #841

closed 2023-10-09 by pascalgouedo