openhwfoundation/cv32e40p #734: Request For contribution

gullahmed1251, 2022-10-16:
Dear Team,

I am motivated by the idea behind open source development and PULP project, so want to contribute to cv32e40p as a project. 
I am adding the support for support for Zba, Zbb, Zbc and Zbs extensions to cv32e40p. I want to start my journey by adding support for Bit Manipulation Extension. Nevertheless, first, I want to understand whether is it really necessary and useful for cv32e40p, or is it something extra that you don't want to consider for now?
I am presenting the proposal presentation to my university tomorrow. I am a final year student of electrical engineering department.
Looking forward for your generous response.

MikeOpenHWGroup, 2022-10-16:
Hi @gullahmed1251, thanks for your interest in CORE-V.  As you probably know, the CV32E40P supports PULP bit-manip as extended ISA instructions.  The on-going CV32E40P v2.0.0 project is currently working to complete the design and verification of these instructions.  Adding support for the RISC-V bit-manip ISA is currently considered a [potential enhancement](https://github.com/openhwgroup/core-v-docs/blob/master/program/Project-Descriptions-and-Plans/CV32E40Pv2/Project_Concept_for_CV32E40Pv2_June-28.md#potential-future-design-enhancements).

I would recommend that you or your Instituion reach out to the [OpenHW Group](https://www.openhwgroup.org/) directly to discuss how to bring your contribution forward for consideration.  Try one of {rick, duncan} at openhwgroup dot org.

One last thing: both the [CV32E40X](https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/index.html) and [CV32E40S](https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/intro.html) support bit-manip, have you looked at these core?

gullahmed1251, 2022-10-16:
Thank you for your generous and quick response. I have checked that CV32E40X and CV32E40S both support bit-manip.
I have read the details that in ALU there are only few instructions for bit-manip. We are planning to add all the instructions in the subgroups Zba, Zbb, Zbc, Zbs extensions in RISCV bit-manip ISA. and I want to know that Is CV32E40P parametrized? means should we add the instructions that supports RV64 bit only?

MikeOpenHWGroup, 2022-11-09:
> I have read the details that in ALU of CV32E40P there are only few instructions for bit-manip.

You would find it a lot easier to read the [User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/index.html).  :stuck_out_tongue_winking_eye:  You can also read about the parameterization that the CV32E40P supports and how to set these at compile/synthesis time.

> We are planning to add all the instructions in the subgroups Zba, Zbb, Zbc, Zbs extensions in RISCV bit-manip ISA.

This could be a very valuable contribution!  None of the current projects within the OpenHW Group anticipate adding Zb* to CV32E40P, but we would consider a serious offer of a contribution to do so.  Please reach out to one or more of {rick, davide, mike} at openhwgroup dot org to discuss.

gullahmed1251, 2023-01-22:
Thank you for your generous response. Yeah, sure this is the serious offer of contribution from our side. We have opted this task as our final year project. Currently we are trying to run the core on verilator along with the FPGA. we are also studying the micro architecture of the core along with the ISA of RISCV. We are very glad and hopeful to complete this project by the end of June, Hopefully.

pascalgouedo, 2023-01-23:
Hi @gullahmed1251 
As you certainly know, CV32E40P is being updated to a v2 version with all PULP instructions being re-encoded and verified to an industrial grade level. This is due to end of June 2023.
For that the decoder RTL code will be quite modified and the verification methodology moved to a state-of-the-art level thanks to Imperas DV methodology. For academics and student projects, I know that Imperas can deliver free licences of its reference model but this is working only in UVM SystemVerilog test-bench, not Verilator for sure.
Concerning adding Zb* to CV32E40P, for sure you can dot it by adding a new parameter like existing PULP_ZFINX which will be renamed ZFINX (e.g. ZBITMANIP).
To anticipate, remember that a (successful) Logic Equivalence Check should be run between frozen v1 RTL version (PULP, CLUSTER, FPU, ZFINX disabled) and any other RTL code with all parameters disabled. I would even add between v2 different versions (PULP=1, CLUSTER, FPU, ZFINX) and any other RTL code with all new parameters disabled (ZBITMANIP).
Cheers,
Pascal.

gullahmed1251, 2023-01-23:
Thank you for your kind words, Sure I am on it now. It will be a great journey. Can you provide me any further information  or link about Logic Equivalence Check(how to do it).

pascalgouedo, 2023-01-23:
https://github.com/openhwgroup/cv32e40p/tree/master/scripts/lec

pascalgouedo, 2023-01-23:
Hi @gullahmed1251 
Just to better know each others, from which country and electrical engineering school/university/... are you?
If you want you can find some info about @MikeOpenHWGroup, @davideschiavone or myself on linkedin...

gullahmed1251, 2023-01-23:
Hey, thanks for asking. I am from Namal University, Pakistan. I am currently in my seventh semester of Electrical Engineering.

gullahmed1, 2023-03-07:
Hi there! I hope you're doing well. I'm currently working on adding bit-manipulation support to the cv32e40p core, and I've just achieved my first milestone of successfully running a basic "hello-world" program using Verilator. 
However, I'm having some trouble simulating the core using ModelSim, as I'm getting an "/'unidentified error." After reading through the documentation, I've decided to try simulating using Verilator instead. I've spent a lot of time learning and understanding Verilator, but I still need some help to save time and get started on writing the RTL for the project. 

Can you provide some details on how I can **simulate the core using Verilator**, or suggest any other simulators? 

Thanks for your time!

MikeOpenHWGroup, 2023-03-07:
@gullahmed1251, @gullahmed1, please reach out to me via email at mike@openwhgroup.org.  I think we need to meet in person to move this discussion forward.

pascalgouedo, 2024-03-18:
As this task of adding Zb extensions will be managed in the scope of [CV32E40PX ](https://github.com/openhwgroup/programs/blob/master/Project-Descriptions-and-Plans/CV32E40PX/Project_Concept_for_CV32E40PX_Nov-27_2023.md) project, I close this issue.

closed 2024-03-18 by pascalgouedo