openhwfoundation/cv32e40p #748: More undefined conditions for HW loops

MikeOpenHWGroup, 2022-12-19:
Reposing here for @tzwaenn

There are various conditions listed in the user manual on undefined behavior for hardware loops for CV32E40P. Looking at this during some experiments for hardware loop support for the Siemens RISC-V verification app with formal verification, some more potentially undefined cases were identified that no sane compiler will generate:

* do not wrap loop bodies (so start address >= end address)
* no self-modifying loops (so setting any register of hwloop x during and iteration of loop x)
* properly entering the loop by executing the instruction at the start address (alternatives are setting count to a value >=2 at a PC location between loop start and end; similarly, setting up all the loop registers, then branch/jump to a location between start and end)

tzwaenn, 2022-12-20:
Thanks for posting in the right place. I was trying to follow the instructions shown when clicking on new issue: "Please file bugs related to the CV32E40P User Manual in https://github.com/openhwgroup/core-v-docs instead"

pascalgouedo, 2022-12-20:
Hi Sven,

core-v-docs has been renamed programs and does not contains Cores user manuals any more.
I think the programs repo title should be renamed as well:
"Documentation for the OpenHW Group's set of CORE-V RISC-V cores" is not totally true now.

Related to the template you are talking about, I am going to make a push request to correct it. Thanks.

Cheers,
Pascal.

davideschiavone, 2022-12-23:
Closing as it seems resolved now - feel free to reopen if needed

closed 2022-12-23 by davideschiavone