openhwfoundation/cv32e40p #756: Blocking assignment in an always_latch block

MikeOpenHWGroup, 2023-01-20:
As discussed in #753, [cv32e40p_register_file_latch.sv](https://github.com/openhwgroup/cv32e40p/blob/132e712c3f7d55c21f831c0c927f5b782391062e/rtl/cv32e40p_register_file_latch.sv#L176) uses a blocking assignment in an `always_latch` block, and I believe that is not compliant with the LRM (clause 9.2.2.3 of 1800-2017).

It is not clear to me that this has caused any functional issues in either RTL or gate-sims.

davideschiavone, 2023-01-23:
as cv32e40p supports at TRL5 only the FF-based RF I added the WAIVED label, but still to be fixed in my opinion. @pascalgouedo ? what do you think? will you use the latch-based RF in your project?

pascalgouedo, 2023-01-24:
Hi,
We are not using latch-based design.
But in my opinion we should correct it and cv32e40p_sim_clock_gate as well to totally get rid of further problems or remarks.

Even if as said by @MikeOpenHWGroup Verilator is not an OpenHW target, it is heavily used by Embecosm to verify all new v2 features for next GCC toolchain.

Cheers,
Pascal.

davideschiavone, 2023-01-24:
the clock gate file should be correct (it uses `<=` and not `=`)

For Verilator we can always apply a compile time waiver

gullahmed1251, 2023-01-24:
I also faced the same non-blocking statement error. I just removed the blocking statement and now it's compiling without any error.

MikeOpenHWGroup, 2023-01-24:
Hi @gullahmed1251, I hope you meant to say "I just replaced the `<=` with `=`".

gullahmed1251, 2023-01-25:
Yeah, you are right. I just noticed it.

MikeOpenHWGroup, 2024-03-22:
Hi @davideschiavone and @pascalgouedo, what is the status of this issue.  I _believe_ it was agreed to fix it, but there hasn't been a resolution and the "waived" label was still applied.

I have removed the label and assigned the issue to both of you.

closed 2024-04-09 by pascalgouedo