openhwfoundation/cv32e40p #757: Extension 'zicsr' required

gullahmed1251, 2023-01-22:
I am trying to run the core by following the [CORE-V-VERIF Verification Strategy (https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html) guide. I have followed all the mentioned steps but when i am finding an error that extension 'zicsr' not found. I am also finding few other errors too.
Kindly check the attached screenshot and help me to solve the errors.
[image]

abukharmeh, 2023-01-22:
It looks like the toolchain you are using wants you to explictly add ziscr to the march used to compile the BSP ! 
So maybe change rv32imc to rv32imc_zicsr in https://github.com/openhwgroup/core-v-verif/blob/0c772a395afb07bdd46aa98ed0ccb5720a99b72b/cv32e40p/bsp/Makefile#L9 ? 

IIRC, zicsr was factored out from the baseline ISA at some point https://gcc.gnu.org/pipermail/gcc-patches/2020-November/558910.html 

If the current version of corev-gcc has this behaviour, then maybe the makefile should be updated ! 

Also I feel like this issue is better suited  for core-v-verif repo ?

gullahmed1251, 2023-01-22:
Hey, thank you so much for helping me. It worked but I found another issue, could you help me in it? This again showing that missing zicsr
[image]

MikeOpenHWGroup, 2023-01-22:
Hi @gullahmed1251, thanks for your issue.  @abukharmeh is correct on both points:

1. You need an up to date toolchain that supports Zicsr.  Prebuilt toolchains from Embecosm.
2. This is not a CV32E40P issue, it is a CORE-V-VERIF issue.

I'll close this and if you are still having trouble after installing an up to date toolchain, please open an issue on CORE-V-VERIF.

closed 2023-01-22 by MikeOpenHWGroup

MikeOpenHWGroup, 2025-02-28:
Hi @aWZHY0yQH81uOYvH, this issue is closed, please do not comment further here.  The discussion has been moved to https://github.com/openhwgroup/core-v-verif/issues/1594.