openhwfoundation/cv32e40p #761: Do u guys have any testbenches for modelsim?

joaopmarinho, 2023-01-29:
I was trying to make some manual tests on modelsim, because apparently there's no test on the platform. Then I'd like to know if you have any version or a complete testbench for the project that runs in modelsim platform.

MikeOpenHWGroup, 2023-01-29:
Hi @joaopmarinho.   The short answer to your question is "probably not".

The long answer goes like this: Modelsim is a SystemVerilog simulator that supports a subset of the SystemVerilog Language Reference Manual (IEEE-1800-2017).  At OpenHW, we exploit most features of SystemVerilog, so you will almost certainly need a simulator such as Siemens Questasim, Cadence Xcelium or Synopsys VCS to simulate our IP.

A short discussion of the OpenHW Group's position on EDA tools is [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/intro.html#a-note-about-eda-tools)

Having said that, the CV32E40P "core" testbench in the CORE-V-VERIF program is known to work with `Verilator`, an open-source SystemVerilog simulator that also supports a subset of the SV LRM.  We would be highly appreciative of a pull-request to add Modelsim 18.1 support.  Check it out at the CORE-V-VERIF [Quick Start Guide](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html).

joaopmarinho, 2023-01-30:
Thankss, I'll be leaving my contributions soon!!!

closed 2023-10-10 by pascalgouedo