openhwfoundation/cv32e40p #782: Does Prefetch Buffer's localparam FIFO_DEPTH  and LSU's localparam DEPTH configurable?

spzeno, 2023-03-03:
[image]

[image]

In cv32e40p_load_store_unit.sv a localparam called DEPTH is found. It was originally 2 and I tried to change this parameter to 8.
In cv32e40p_prefetch_buffer.sv a localparam called FIFO_DEPTH is found. It was originally 2 and I tried to change this parameter to 8.
But I found that the CPU data channel would make an error when running the following program:
[image]

[image]

The "sw	zero,48(a0)" request is repeated multiple times.
Is this because these two parameters cannot be changed at will?

Looking forward to reply!

pascalgouedo, 2023-03-03:
Hi
LSU localparam DEPTH shouldn't be changed because there is nothing which is memorizing outstanding access information for different load instructions to be used when corresponding response will come (like GPR number to be updated or load type).
The only case when 2 outstanding requests could be sent away is for 1 misaligned load where both data will be merged to write the destination GPR.

On Prefetch Buffer side, localparam FIFO_DEPTH can be changed to any power of 2 value.

spzeno, 2023-03-03:
> Hi LSU localparam DEPTH shouldn't be changed because there is nothing which is memorizing outstanding access information for different load instructions to be used when corresponding response will [...]

Thanks very much for your reply!

closed 2023-03-03 by spzeno