openhwfoundation/cv32e40p #784: Different bahavior between vsim and verilator for mcycle csr

Lucas-Wye, 2023-03-08:
When I simulate with verilator, I can get the `correct cycle number` using `rdcycle` instruction.
However, if I use vsim, I always get `cycle` = 0.

I think this may because of different initializations of reg in verilator and vsim.

See here:
https://github.com/openhwgroup/cv32e40p/blob/2d59dae2114b004127e1f2dbb9ae40fc0961085e/rtl/cv32e40p_cs_registers.sv#L1494
https://github.com/openhwgroup/cv32e40p/blob/2d59dae2114b004127e1f2dbb9ae40fc0961085e/rtl/cv32e40p_cs_registers.sv#L1495

Because `mcountinhibit_q[0]` is initialized as `1`,  `mcountinhibit_n[0]` would always be `1`.
However, in verilator, `mcountinhibit_n[0]` is initialized as `0`. 
I am not sure if I am correct.

To address the problem of wrong cycle number in `vsim`, I suggest to change line `1494` to:
```verilog
          if (!rst_n) mcountinhibit_q[inh_gidx] <= 'b0;  // default enable
```

MikeOpenHWGroup, 2023-03-08:
Hi @Lucas-Wye. Thanks for your interest in CV32E40P.   Can you provide more information?  I specifically need to know:
1. The _exact_ command-lines you use to reproduce this issue, starting from `git clone...`.
2. The version of Questasim you are using.
3. The version of Verilator you are using.

pascalgouedo, 2023-03-08:
Hi @Lucas-Wye 

I am using vsim and don't experience your always at 0 read pb.
I used csr instructions to read either cycle or mcycle CSR as well as rdcycle instruction.
For sure after having enabled cycle counter using:
`  __asm__ volatile("csrci mcountinhibit, 0x1"); // mcountinhibit.cy = 0`

Anyway from a design point of view we can't take your suggestion because it would enable by default counters and this is bad for power consumption.

Lucas-Wye, 2023-03-08:
> Hi @Lucas-Wye
>
> I am using vsim and don't experience your always at 0 read pb. I used csr instructions to read either cycle or mcycle CSR as well as rdcycle instruction. For sure after having enabled [...]

Thank you so much. Here is an example of how I use `rdcycle`:
```c
static uint32_t rdcycle_read()
{
    register uint32_t num_cycles;
    __asm__ volatile("rdcycle %0;" : "=r"(num_cycles));
    return num_cycles;
}

int main(void) {
    uint32_t cycle_start = rdcycle_read();

    // some operations ...

    uint32_t cycle_end = rdcycle_read();
    printf("#cycle = %lu\n", cycle_end - cycle_start);
}
```
Maybe there is something wrong because initially I use `verilator` and it can always give me the cycles.
Do you mean I should run 
```
  __asm__ volatile("csrci mcountinhibit, 0x1"); // mcountinhibit.cy = 0
```
before `rdcycle`? Or could you give me some examples? Thank you.

pascalgouedo, 2023-03-08:
Yes you need to enable counter to let them count (RISC-V spec) and then you can read their values.
If you don't do that you will always read reset value meaning 0.

I don't use Verilator so I don't know why it has an incorrect behaviour ...

Lucas-Wye, 2023-03-08:
> 

Thank you so much.  Also thanks to @MikeOpenHWGroup.
I think now it is just an issue of verilator.

MikeOpenHWGroup, 2023-03-08:
Hi @Lucas-Wye, if this issue is resolved, please close it.  Thanks!

closed 2023-03-08 by Lucas-Wye