openhwfoundation/cv32e40p #790: Where is Pulp_Clocking_Gating file?

BhattSoham, 2023-04-08:
HI there,
       Good day. I was trying to run this core wrapper system Verilog files in vivado environment and trying to synthesize it. During those steps, I am not able to find the pulp_clock_gating files and also the branch files of the clock divider. Could you please help me out finding those sources?
       
Thanks in advance,

MikeOpenHWGroup, 2023-04-09:
Hi @BhattSoham, thanks for your interest in the CV32E40P!   Have you looked at the Clocking Gating section of the [User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/getting_started.html#clock-gating-cell)?

BhattSoham, 2023-04-09:
Yes I think I'll find it out through the clock gating section. Just to confirm some things, the cv32e40p is the original file of the riscv core, right? Actually, I am trying to synthesize that using vivado. One more thin I need to know, if I want to do some operations using some instructions, is there any specific procedure for that?

MikeOpenHWGroup, 2023-04-09:
Sorry @BhattSoham, the CV32E40P project does not explicitly support Vivado.  If a new or existing member of the OpenHW Group wanted to contribute Vivado support, that would be great.  If you are interested in that, please reach out to Davide Schiavone, our Director of Engineering for the Cores Task Group.

You could also look at [CORE-V-MCU](https://github.com/openhwgroup/core-v-mcu) which integrated CV32E40P into an MCU and does have Makefiles for Vivado.

Since your original question has been answered, I will close this Issue.  We do not like to have multiple topics in a single Issue.

closed 2023-04-09 by MikeOpenHWGroup