openhwfoundation/cv32e40p #803: genus reports referenced package cv32e40p_fpu_pkg does not exist

Ceridli, 2023-05-04:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

**Component:Tool-and-build**: For issues in the tool and build flow (e.g. Makefile, FuseSoc, etc.)

### Steps to Reproduce
Please provide

1. git hash : I don't know what this is, but I used version  1.3.0
3. Command line:  I used 'bender script genus > bender-genus.tcl'  at the top of source directory to generate synthesis scripts.
4. Logfile and/or wave-dump info (screen shots can be useful)

Setting attribute of root '/': 'init_hdl_search_path' = /apps1/customers/rtl/cv32e40p-cv32e40p_v1.3.0/.bender/git/checkouts/common_cells-65dd26b9bd97b679/include /apps1/customers/rtl/cv32e40p-cv32e40p_v1.3.0/rtl/include
@file(bender-genus.tcl) 243: read_hdl -language sv \
    -define { \
        TARGET_GENUS \
        TARGET_SYNTHESIS \
    } \
    [list \
        "$ROOT/rtl/include/cv32e40p_apu_core_pkg.sv" \
        "$ROOT/rtl/include/cv32e40p_pkg.sv" \
        "$ROOT/rtl/cv32e40p_alu.sv" \
        "$ROOT/rtl/cv32e40p_alu_div.sv" \
        "$ROOT/rtl/cv32e40p_aligner.sv" \
        "$ROOT/rtl/cv32e40p_compressed_decoder.sv" \
        "$ROOT/rtl/cv32e40p_controller.sv" \
        "$ROOT/rtl/cv32e40p_cs_registers.sv" \
        "$ROOT/rtl/cv32e40p_decoder.sv" \
        "$ROOT/rtl/cv32e40p_int_controller.sv" \
        "$ROOT/rtl/cv32e40p_ex_stage.sv" \
        "$ROOT/rtl/cv32e40p_hwloop_controller.sv" \
        "$ROOT/rtl/cv32e40p_hwloop_regs.sv" \
        "$ROOT/rtl/cv32e40p_id_stage.sv" \
        "$ROOT/rtl/cv32e40p_if_stage.sv" \
        "$ROOT/rtl/cv32e40p_load_store_unit.sv" \
        "$ROOT/rtl/cv32e40p_mult.sv" \
        "$ROOT/rtl/cv32e40p_prefetch_buffer.sv" \
        "$ROOT/rtl/cv32e40p_obi_interface.sv" \
        "$ROOT/rtl/cv32e40p_core.sv" \
        "$ROOT/rtl/cv32e40p_apu_disp.sv" \
        "$ROOT/rtl/cv32e40p_fetch_fifo.sv" \
        "$ROOT/rtl/cv32e40p_popcnt.sv" \
        "$ROOT/rtl/cv32e40p_ff_one.sv" \
        "$ROOT/rtl/cv32e40p_sleep_unit.sv" \
    ]
  import cv32e40p_fpu_pkg::*;
                        |
Error   : The referenced package does not exist.. [VLOGPT-28] [read_hdl]
        : Reference to non-existent package 'cv32e40p_fpu_pkg' in file '/apps1/customers/rtl/cv32e40p-cv32e40p_v1.3.0/rtl/cv32e40p_decoder.sv' on line 31, column 25.
        : The referenced package does not exist in the same HDL library.
  output logic [cv32e40p_fpu_pkg::FP_FORMAT_BITS-1:0]  fpu_dst_fmt_o,   // fpu destination format
                                 |
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Bad or unsupported package prefix 'cv32e40p_fpu_pkg' in file '/apps1/customers/rtl/cv32e40p-cv32e40p_v1.3.0/rtl/cv32e40p_decoder.sv' on line 111, column 34.
        : Invalid Verilog syntax is parsed, or unsupported Verilog syntax is encountered.

pascalgouedo, 2023-05-04:
Hi @Ceridli,

Bender.yml and src_files.yml files are not used in OpenHW group methodology, only cv32e40p_manifest.flist.
Those yaml files came from ETHZ repo and I am not sure they have been used/updated for v1.0.0 RTL freeze.
What can be seen is that v1.0.0 version of cv32e40p_decoder.sv was already using that fpu package but Bender file was not listing it.

For sure between v1.0.0 and v1.3.0 I have not updated them, only updated cv32e40p_manifest.flist and added another one including FPU called cv32e40p_fpu_manifest.flist
So Bender file including FPU does not exist even if I see dependencies to fpnew git repos (now managed with local copies using vendorization flow).

You can submit a PR to dev only touching those files to fulfil your needs, we will merge it in dev and master after review.

Cheers,
Pascal.

davideschiavone, 2023-10-10:
@Ceridli , feel free to send a PR with the fixes for Bender if you wish, happy to merge them