openhwfoundation/cv32e40p #814: xpulp SIMD ALU instruction - CV.AVGU.SCI 6-bit Immidiate should be zero extended in the User Manual

dd-vaibhavjain, 2023-05-30:
### Issue Description

**Section 17.6.1 SIMD ALU operations in User-manual-en-cv32e40p_v1.3.0**  
For cv.avgu instruction the Description does not mention if Immediate is zero-extended, and as mentioned in section 17.6 above in this user manual, in such cases the Immidiate is considered as sign-extended.

But in RTL it is not sign extended and considered as zero-extended. Though RTL seems not aligned to spec, looking at other similar instructions in this extension like cv.minu. cv.maxu and the difference between cv.avg and cv.avgu being that cv.avgu has logical right shift, i think this is probably a User Manual/Doc bug leading to this issue. 

We got this issue in a failing test where the reference model considered this Imm value as sign-extended (matching spec currently by default) but RTL behaving differently :  
Info (IDV) Instruction executed prior to mismatch '0x35e(main+394): **1936e17b cv.avgu.sci.h x2,x13,-26'**  
Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000035e main+394):  
Info (IDV)  0> GPR x2  
Info (IDV)   . **dut:0x00130013**  
Info (IDV)   . ref:0x7ff37ff3  

### Component

**Component:Doc**  

### Steps to Reproduce  

1. GIT hash for cv32e40p and core-v-verif  
    cv32e40p Hash : f789eb83bce527664867cae9d21a584b28492935 ;  
    core-v-verif ; branch - cv32e40p/dev ; hash : f4a83bd3ffc9ac69584f17b9d13fa0d77b22c833  

2. run commands:   
    cd core-v-verif/cv32e40p/sim/uvmt  
    make clean gen_corev-dv TEST=corev_sanity_instr_test CFG=pulp_fpu SEED=906506742 SIMULATOR=vsim  
    make test TEST=corev_sanity_instr_test USE_ISS=yes WAVES=1 CFG=pulp_fpu SEED=521863613 SIMULATOR=vsim

pascalgouedo, 2023-05-30:
Will add a comment on cv.avgu anyway.

dd-vaibhavjain, 2023-05-31:
> In this case operand sign depends of the instruction itself. avg treats signed operands and avgu unsigned ones. So it seems quite straightforward that Imm6 is either signed or unsigned, not depending on a general statement at the beginning of the section.

It is ambiguous, more so because cv.maxu and cv.minu in this same SIMD ALU Instructions section have explicit Note for Imm6 being Zero-Extended, and this specific Note is not mentioned for cv.avgu, so this creates ambuiguity considering that within the same section other similar instructions have a clear Note along with general statement above. 
So while it can be assumed the cv.avg and cv.avgu behaviour as implemented in RTL is correct, it is definitely not clearly defined in the User Manual which leads to this issue.  
We also needs this clarity in Spec to justify the fix in reference model.

dd-vaibhavjain, 2023-07-13:
Thanks @pascalgouedo !
This issue is addressed and User Manual updated to indicate "Immediate is zero-extended, shift is logical".
Closing this issue.

closed 2023-07-13 by dd-vaibhavjain