openhwfoundation/cv32e40p #817: The encoding table for the bit manipulation extension is not rendered

melonedo, 2023-06-14:
# Issue Description

[image]

The [encoding table](https://github.com/openhwgroup/cv32e40p/blob/master/docs/source/instruction_set_extensions.rst#bit-manipulation-encoding) for the bit manipulation extension is not rendered. There is source code for this table but it is not rendered in GitHub preview. https://github.com/openhwgroup/cv32e40p/blob/0391e5c15aa32a3df37179ac1d9a17f5f76cbc1e/docs/source/instruction_set_extensions.rst?plain=1#L648-L705

### Component

**Component:Doc**

pascalgouedo, 2023-06-14:
Hi @melonedo 
Thanks for figuring out this. There was 1 remaining "width" command on those 2 encoding tables.
I removed them locally and saw that it restored github rendering while leaving generated sphynx html and pdf the same as before.
Will commit my update.

pascalgouedo, 2023-08-04:
Corrected with PR #818

closed 2023-08-04 by pascalgouedo