openhwfoundation/cv32e40p #831: What is the current status of the standard?

ChunyuLiao, 2023-07-10:
If not frozen, is there a plan for when it will be frozen?

We are submitting patch to [upstream llvm](https://github.com/openhwgroup/corev-llvm-project/issues/65) and need to know the current status of the standard.

MikeOpenHWGroup, 2023-07-11:
Hi @ChunyuLiao, thanks for your interest in CV32E40P.   What "standard" are you asking about?  The XPULP ISA?  Probably the best person to comment on this is @pascalgouedo, but he is on personal leave for a couple of weeks.  @jeremybennett can you comment?

ChunyuLiao, 2023-07-12:
thanks@MikeOpenHWGroup
I'm asking about CV32E40P.
As a standard extension. there are [five Specification States](https://wiki.riscv.org/display/HOME/Specification+States).

our CV32E40P, as a vendor extension,   is there a plan for when it will be frozen.

MikeOpenHWGroup, 2023-07-12:
Please note that the XPULP ISA is _not_ a RISC-V standard ISA and does not follow the "five states" of a RISC-V International specification.  Rather, it is a custom vendor extension that is explicitly tied to the implementation of the CV32E40P.  So we can consider the XPULP ISA to be frozen when release v2.0.0 of the CV32E40P is made.  I believe the goal for this is October of this year.

I am aware that there has been some discussion about creating a more formal specification (beyond what is published in the CV32E40P User Manual), but do not know if these discussions are moving forward.  If you are interested in seeing this happen, you could consider joining the OpenHW Group and making a proposal.

ChunyuLiao, 2023-07-13:
Thanks, looking forward to the CV32E40P freezing soon.

MikeOpenHWGroup, 2023-07-13:
Hi @ChunyuLiao, if I have answered your question, please close this issue.  Thanks!

jeremybennett, 2023-07-14:
@MikeOpenHWGroup I think this answered the wrong question from @ChunyuLiao .  This is the specification of the core.  From a compiler perspective it is the numbering of the versions of the ISA which matters for upstream acceptance.  This was frozen earlier this year (see [corev-isa-extension-naming.md](https://github.com/openhwgroup/core-v-sw/blob/master/specifications/corev-isa-extension-naming.md)).

The tool chain is not supporting versioning of the ISAs to be supported (which is going to become the standard RISC-V builtins). 
 The only ISA extensions we shall support are those added to CV32E40Pv2.  Since there is only one version of the ISA extensions supported in the tool chains, they must be (per RISC-V standards) ISA extension version 1.0.0.

I'll submit a patch to the documentation of the ISA extensions in the main document to make it clear that the ISA extension version from a compiler perspective is 1.0.0.  This has caused a disproportionate amount of review feedback in the upstream Clang/LLVM submission.

jeremybennett, 2023-07-14:
Closing again after adding the supplementary comment.

closed 2023-07-14 by jeremybennett