openhwfoundation/cv32e40p #833: Naming of cv.slet

realqhc, 2023-07-14:
On a review for upstreaming core-v alu encoding, the reviewer Craig Topper asked

> I would have expected cv.sle and cv.sleu for these mnemonics. The t feels out of place. In the base isa for slt/sltu it's part of "less than". Is the instruction name here "set less equal than" instead "set less than or equal"?
> 
> The vector spec uses "sle" and "sleu"

Is it possible that anyone could provide a clarification on the meaning of the mnemonics cv.slet, cv.sletu?

pascalgouedo, 2023-07-17:
Hi @realqhc 

Those instructions are executing a "set if less than or equal" operation.
So I agree that cv.sle/cv.sleu would be a more standard way of mnemonic naming.
It has been inherited from PULP RI5CY and I didn't figured out this point before.

As PULP instructions are close to frozen (especially in GNU toolchain), I will ask to all concerned people if renaming is still possible at this point of the project.

Thank you anyway for your comment/issue.

Best regards,
Pascal.

davideschiavone, 2023-08-04:
@realqhc , can we close this issue?

pascalgouedo, 2023-08-04:
Only once #844 merged ...

closed 2023-08-04 by realqhc