openhwfoundation/cv32e40p #838: pulp hwloop count not decrementing to 0 after the end of hwloop body execution

dd-vaibhavjain, 2023-07-25:
### Issue Description
In pulp hwloop tests, after the end of hwloop body, i.e., after the last hwloop body instruction has been executed, the loop count csr does not decrement to 0 and final value of count remains at 1

### Component
**Component:RTL**

### Steps to Reproduce

1. cv32e40p git hash : 99772238edb4093e99386599b9f3d79f4d125f30

2.  Using "core-v-verif" repo ,  branch : cv32e40p/dev
     core-v-verif hash :  c7610d8ab6fecdb81b21a6e4e791c8f9e01fc7bd
     
-      git clone -b cv32e40p/dev https://github.com/openhwgroup/core-v-verif.git
-      cd core-v-verif
-      git checkout c7610d8ab6fecdb81b21a6e4e791c8f9e01fc7bd
  
     
3. Run Test:  
Run with reference model (USE_ISS=yes) to get the error mismatch in log. Otherwise if running without USE_ISS, we can see this issue from the wave dump.
-     setenv SIMULATOR vsim (or any other simulator)
-     cd cv32e40p/sim/uvmt/
-      make test TEST=pulp_hardware_loop USE_ISS=yes CFG=pulp_fpu WAVES=1 SEED=1 RUN_INDEX=1 USER_RUN_FLAGS="+IDV_TRACE2LOG=1"

4. Log Error:

> Info (IDV) Instruction executed prior to mismatch '0x1c0(startZ_1+8): 00188893 addi    x17,x17,1'
> Error (IDV) CSR register value mismatch (HartId:0, PC:0x000001c0 startZ_1+8):
> Info (IDV)  0> CSR cc2 (lpcount0)
> Info (IDV)   . dut:0x00000001
> Info (IDV)   . ref:0x0000000

davideschiavone, 2023-08-03:
can we close this as now 788caae7d5e4d5fc07f68dc7f34f6307ca17f5d4 has been merged?

dd-vaibhavjain, 2023-08-04:
Thanks @pascalgouedo for this fix. 
Verified the RTL fix and it resolves the issue here. RTL and ISS model both are aligned for hwloop tests. 
Closing this ticket.

closed 2023-08-04 by dd-vaibhavjain