openhwfoundation/cv32e40p #839: Link to readthedocs on landing page no longer works

jstraus59, 2023-07-27:
As of today the [link to the documentation on the landing page](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest) has stopped working.

This is what I get when I select it: 
[image]

It was working yesterday. Looks like something changed at readthedocs?

jstraus59, 2023-07-27:
FYI, using google I did find this alternative URL that does work:
https://openhw-group-cv32e40p-user-manual.readthedocs-hosted.com/en/cv32e40p_v1.3.2

MikeOpenHWGroup, 2023-07-27:
Hi Jim.  This is a fat-finger problem at our end.  It is being actively worked on.  In the meantime, for the E40P, the following link works: https://cv32e40p.readthedocs.io/en/latest/index.html.

jstraus59, 2023-07-27:
Thanks Mike. Closing issue.

closed 2023-07-27 by jstraus59