openhwfoundation/cv32e40p #843: Wrong Hierarchical Wiring in cv32e40p_tb_subsystem.sv Line **157**

DerakhshanAli, 2023-08-03:
Bug Title:
Wrong Hierarchical Wiring
in cv32e40p_tb_subsystem.sv Line **157** 
   . . . 
   Line 157:     .pc_core_id_i(**_wrapper_i_**.core_i.pc_id),   
   . . .

    I think The  wrapper_i.core_i.pc_id  must change into a top_i.core_i.pc_id

**Component:RTL**:
cv32e40p_tb_subsystem.sv

cv32e40p-cv32e40p_v1.3.2.

davideschiavone, 2023-08-11:
Hello @DerakhshanAli and thanks for sharing this issue with us.

I do not see this issue here https://github.com/openhwgroup/cv32e40p/blob/master/bhv/cv32e40p_tb_wrapper.sv#L157

can you point me where is there problem? thanks

DerakhshanAli, 2023-08-11:
Glad to help.

In this Directory:
cv32e40p_v1.3.2\example_tb\core

cv32e40p_tb_subsystem.sv Line 157
. . .
Line 157: .pc_core_id_i(wrapper_i.core_i.pc_id),
______________________________________________________________________________
I encountered this error when I wanted to Run Core simulation according to Makefile in cv32e40p_v1.3.2\example_tb\core directory with [make custom-vsim-run-gui] Command.

davideschiavone, 2023-08-11:
ah sorry  I looked at the wrong file, the issue is here https://github.com/openhwgroup/cv32e40p/blob/master/example_tb/core/cv32e40p_tb_subsystem.sv#L157, agree - @pascalgouedo , can we fix this in the next release?

DerakhshanAli, 2023-08-11:
Glad to contribute.

In this Directory:
cv32e40p_v1.3.2\example_tb\core

cv32e40p_tb_subsystem.sv Line 157
. . .
Line 157: .pc_core_id_i(wrapper_i.core_i.pc_id),

I encountered this issue when I wanted to Run Core simulation according to Makefile in cv32e40p_v1.3.2\example_tb\core    directory with   [make custom-vsim-run-gui]   Command.

pascalgouedo, 2023-08-17:
Fixed with #851.

closed 2023-08-17 by pascalgouedo