openhwfoundation/cv32e40p #850: restrictions in HWLoop

davideschiavone, 2023-08-11:
In the user manual https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/corev_hw_loop.html, it is written

`
No privileged instructions (mret, dret, ecall, wfi) allowed in the HWLoop body, except for ebreak.
`

However, in the RTL it is written that such a condition is possible: https://github.com/openhwgroup/cv32e40p/blob/master/rtl/cv32e40p_controller.sv#L788

This applies also to other constraints, for example csr access.

pascalgouedo, 2023-08-11:
To summarize all updates:

- removing is_hwlp_illegal [here](https://github.com/openhwgroup/cv32e40p/blob/5f5fd49bac2db05508ab9c240782fc8c7c742268/rtl/cv32e40p_controller.sv#L530) and [here](https://github.com/openhwgroup/cv32e40p/blob/5f5fd49bac2db05508ab9c240782fc8c7c742268/rtl/cv32e40p_controller.sv#L755) as not complete with respect to HW loops constraints,
- removing ecall_insn_i [here](https://github.com/openhwgroup/cv32e40p/blob/5f5fd49bac2db05508ab9c240782fc8c7c742268/rtl/cv32e40p_controller.sv#L788C21-L788C33),
- removing data_load_event_i [here](https://github.com/openhwgroup/cv32e40p/blob/5f5fd49bac2db05508ab9c240782fc8c7c742268/rtl/cv32e40p_controller.sv#L799).

For csr accesses (related to [csr_status_i](https://github.com/openhwgroup/cv32e40p/blob/5f5fd49bac2db05508ab9c240782fc8c7c742268/rtl/cv32e40p_controller.sv#L794)) I think it should be kept in RTL because of possible FRM change inside a HW loop.
But maybe a constraint should be added to list which CSR accesses (and their R/W type) are authorized inside a loop.

davideschiavone, 2023-08-11:
note that to keep area constraints under control, we must specify what is illegal, but not take real actions (e.g. trapping to an illegal routine) - this is left up to you @pascalgouedo - I would probably add illegals check if the area overhead is small.

Note also that when returning from an interrupt, the CPU can go back to the DECODE state instead of DECODE_HWLOOP, thus those checks need to be propagated there as well (today they are not)

jstraus59, 2023-08-11:
I am implementing the hwloop constraints checks in the Imperas simulation model for the CV32E40P and I have a few questions:

One issue is precisely when these checks occur. You don't want to do any checking until a loop actually executes, i.e. the checks cannot occur when a hwloop CSR is written. (Since updating the loop start/end is not an atomic operation, until all hwloop CSRs are updated a bogus loop range would be checked.) 

I believe the only solution is to do checks when a hwloop implied branch at the end of the loop occurs. At this point we can check:
- the alignment and size of the loop, 
- scan the instructions between the loop start and end for any that violate the rules (e.g. compressed, branch, jump, fence, fence.i, mret, dret, ecall or wfi).

However, I see no way to check the constraint `HWLoop must always be entered from its start location (no branch/jump to a location inside a HWLoop body`, and in fact I wonder if this should really even be a constraint, as I cannot imagine an implementation where this would cause problems. In the simulator the **only** loop instruction that has any extra coding is the last instruction, where the branch implied by the loop is executed. The instruction that matches the loop start and all the instructions in the loop body besides the last are unaffected. 

Is it possible that the HW implementation is currently complicated by initial attempts to implement hardware checks, and now that the decision has been made to remove those checks that a much simpler approach could be used? Or is there some other reason, besides error checking, why the hardware must keep state information indicating when it is executing an instruction in a loop? I would be curious what that reason is, because we certainly aren't tracking that in the simulation model.

Additional questions:
1) Is it legal for the start0 and start1 addresses to be the same (i.e. nested loops with the same start)?
2) May a start address for one loop be the same as the end address of the other loop (i.e. consecutive loops)?
3) For the constraint `When both loops are nested`, what defines when loops are considered nested? The only check I can imagine is that when the (inner) loop0 end address is executed that the loop1 end address is checked that it is not in the range loop0:loop0+8. Is this the correct check?
4) If the last instruction in the loop body is executed and the corresponding loop count CSR is 0 should any checks be performed?
5) What should the loop start/end CSRs be set to when loops are not being used? Setting to 0 seems problematic, since jumping to 0 would then trigger illegal loop errors. 
6) Perhaps setting a loop *count* CSR to a non-zero value should be what triggers the loop checks? Thus setting loop count CSR to 0 deactivates all checks and behavior for the corresponding loop start/end CSRs?

pascalgouedo, 2023-08-18:
>     * removing ecall_insn_i ... 

After a talk with @jeremybennett it is ok to exclude ecall from HW Loop body.

pascalgouedo, 2023-08-18:
Hi @jstraus59 
Starting to answer to your questions...

> One issue is precisely when these checks occur. You don't want to do any checking until a loop actually executes, i.e. the checks cannot occur when a hwloop CSR is written. (Since updating the loop [...]

It depends of the constraint to be checked I guess.
For instance first constraint must be checked when the CSR are written using start, end and setup instructions.
And another bunch of them must be checked when executing first instruction of any HW loop body (rather that at the end as you mentioned because the design could never reach it if there is an undefined behaviour due to the execution of an excluded instruction).
And some of them when executing every instruction inside the body.

pascalgouedo, 2023-08-18:
> However, I see no way to check the constraint HWLoop must always be entered from its start location (no branch/jump to a location inside a HWLoop body, and in fact I wonder if this should really [...]

For this one I agree it is maybe hard to check.
I didn't try this case it is coming from @davide past experience (v1 verification?).
It is maybe caused by side effects related to interrupts and debug?
Anyway I tend to agree that as HW loops are used to make computation loops faster, they are generally quite simple and are not entered by jumps nor contain any one of those jumps.
I will make some trials to see what really happens there.

pascalgouedo, 2023-08-18:
> Is it possible that the HW implementation is currently complicated by initial attempts to implement hardware checks, and now that the decision has been made to remove those checks that a much [...]

HW loop hardware checks were not that much complicated originally.
And yes besides error checking, the design uses that "in hwloop body" info and even has a specific state in its controller state machine used to correctly manage interrupts and debug requests as well as jumping correctly at the end of the loop to avoid to loose cycles we are trying to gain with this mechanism.

pascalgouedo, 2023-08-18:
> Additional questions:
> 
> 1. Is it legal for the start0 and start1 addresses to be the same (i.e. nested loops with the same start)?
    
Not sure about that.
Normally there is at least the inner loop counter initialization in between.
But if this initialization is done once before outer loop and after the end of the inner loop then it could be possible to have start0 = start1.
Will make a try.

pascalgouedo, 2023-08-18:
> 2. May a start address for one loop be the same as the end address of the other loop (i.e. consecutive loops)?

I guess so.
Will try.

pascalgouedo, 2023-08-18:
> 3. For the constraint When both loops are nested, what defines when loops are considered nested? The only check I can imagine is that when the (inner) loop0 end address is executed that the loop1 end address is checked that it is not in the range loop0 end:loop0 end+8. Is this the correct check?

inner loop start address > outer loop start address
inner loop end address < outer loop end address

Like the example given in the [User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/corev_hw_loop.html).

I guess this can only be checked when executing first instruction of the inner HW loop body.

pascalgouedo, 2023-08-18:
> 4. If the last instruction in the loop body is executed and the corresponding loop count CSR is 0 should any checks be performed?

All checks already done before so nothing special here.

pascalgouedo, 2023-08-18:
> 5. What should the loop start/end CSRs be set to when loops are not being used? Setting to 0 seems problematic, since jumping to 0 would then trigger illegal loop errors.

0 is all CSRs reset value (with some exceptions), even HW loop ones.

> 6. Perhaps setting a loop count CSR to a non-zero value should be what triggers the loop checks? Thus setting loop count CSR to 0 deactivates all checks and behavior for the corresponding loop start/end CSRs?

This is for sure. HW loop mechanism only starts comparing PC/Start/End addresses when lpcount0/1 are strictly greater than 0.

jstraus59, 2023-08-18:
> 0 is all CSRs reset value (with some exceptions), even HW loop ones.
>
>
> This is for sure. HW loop mechanism only starts comparing PC/Start/End addresses when lpcount0/1 are strictly greater than 0.
 
So I am thinking now that checks should be triggered each time an address matching a hwloop startX address is executed AND countX is non-zero. Does this sound correct?
If so, then perhaps the documentation should reflect this? It is kind of vague about this - it specifies rules that, for example, are violated when all CSRs are 0, or may be violated between the writing of the startX and endX registers.

pascalgouedo, 2023-08-22:
Some checks could only be done when PC equals lpstartX and lpcountX is different than 0.
Some of them should be done for every instruction when (lpstartX <= PC <= lpendX) and lpcountX is different than 0.
So maybe the simplest is just to have only the second case.

pascalgouedo, 2023-08-22:
> If so, then perhaps the documentation should reflect this? It is kind of vague about this - it specifies rules that, for example, are violated when all CSRs are 0, or may be violated between the writing of the startX and endX registers.

I am going to add a comment about that.

pascalgouedo, 2023-08-22:
> when (lpstartX <= PC <= lpendX) and lpcountX is different than 0.

(lpstartX <= PC <= lpendX **- 4**) in fact.

closed 2023-08-23 by pascalgouedo