openhwfoundation/cv32e40p #895: Hello world simulating optimization failed error

vidushiy25, 2023-10-23:
Hi, I want to use the cv32e40p core for a project and I am just trying to run a simulation for the first time. However it gives me 2 errors : Top level modules:
	tb_top

Analyzing design...
-- Loading module tb_top
-- Loading module cv32e40p_tb_subsystem
-- Loading module cv32e40p_top
-- Importing package cv32e40p_apu_core_pkg
** Error (suppressible): cv32e40p_tb_subsystem.sv(74): (vopt-2732) Module parameter 'PULP_XPULP' not found for override.
** Error (suppressible): cv32e40p_tb_subsystem.sv(75): (vopt-2732) Module parameter 'PULP_CLUSTER' not found for override.
-- Loading module cv32e40p_core
-- Importing package cv32e40p_pkg
-- Loading module cv32e40p_sleep_unit
-- Loading module cv32e40p_clock_gate
-- Loading module cv32e40p_if_stage
-- Loading module cv32e40p_prefetch_buffer
-- Loading module cv32e40p_prefetch_controller
-- Loading module cv32e40p_fifo
-- Loading module cv32e40p_obi_interface
-- Loading module cv32e40p_aligner
-- Loading module cv32e40p_compressed_decoder
-- Loading module cv32e40p_id_stage
-- Importing package cv32e40p_fpu_pkg
-- Loading module cv32e40p_register_file
-- Loading module cv32e40p_decoder
-- Loading module cv32e40p_controller
-- Loading module cv32e40p_int_controller
-- Loading module cv32e40p_ex_stage
-- Loading module cv32e40p_alu
-- Loading module cv32e40p_popcnt
-- Loading module cv32e40p_ff_one
-- Loading module cv32e40p_alu_div
-- Loading module cv32e40p_mult
-- Loading module cv32e40p_load_store_unit
-- Loading module cv32e40p_cs_registers
-- Loading module mm_ram
-- Loading module amo_shim
-- Loading module dp_ram
-- Loading module riscv_rvalid_stall
-- Importing package perturbation_pkg
-- Loading module riscv_gnt_stall
-- Loading module cv32e40p_random_interrupt_generator
Optimization failed
End time: 16:57:29 on Oct 23,2023, Elapsed time: 0:00:00
Errors: 2, Warnings: 1
make: *** [.opt-rtl] Error 2
[Makefile.zip](https://github.com/openhwgroup/cv32e40p/files/13073156/Makefile.zip)

All I did was remove the VVERSION suffix in my makefile and added the riscv toolchain path.

MikeOpenHWGroup, 2023-10-23:
Hi @vidushiy25.  Thanks for your interest in OpenHW.  Can you share all the shell commands you used, starting with `git clone...`.  A couple of other things to point out:
- it seems you are running MACOSX. AFAIK, our scripts have never been tested on that OS.
- this "example-tb" is not the one we use for verification of this core.  For that, check out [core-v-verif](https://github.com/openhwgroup/core-v-verif/).  If you are interested in the CV32E40P, then I suggest you work with the `cv32e40p/dev/ branch of core-v-verif.

vidushiy25, 2023-10-23:
Hi, I am using NoMachine to log onto a Windows desktop. Regardless, I did get the simulation running but yes I did find out that this example_tb is not the one to use for this core. I will be using what you pointed out. Thanks

closed 2023-10-23 by vidushiy25

MikeOpenHWGroup, 2023-10-23:
> I am using NoMachine to log onto a Windows desktop

I am not aware of anyone who has successfully used our IP and scripts on a Windows platform.  Not recommended.

vidushiy25, 2023-10-23:
Sorry, I meant I am using NoMachine to access the Linux desktop not Windows.