openhwfoundation/cv32e40p #943: Error in Anakysis and Synthesis in Quartus

vidushiy25, 2024-02-02:
Hi, I'm trying to synthesise the core and its soc design on an Altera FPGA. But during the Analysis and Synthesis, it's giving me syntax errors in various source code files like the id_stage.sv, ex_stage.sv. The issue seems to be mostly coming from files which are importing packages and in some the syntax errors are deep within the source files like a variable being outside the scope. I have prioritised packages first in the list but it didn't make any difference. 

I included all packages and files for the core in the same directory for Quartus synthesis. 
Please advice. 
[image]
[image]

pascalgouedo, 2024-02-02:
This RTL is correctly simulated by all 3 Siemens, Cadence and Synopsys simulators.
It is implemented using Synopsys Design Compiler without any troubles.
And FPGA mapping for AMD/Xilinx Artyx 7 using vivado works fine as well.
The pb surely comes from your Quartus setup.
Include paths mentioned in cv32e40p_manifest.flist and cv32e40p_fpu_manifest.flist should be added in simulation or implementation tool setup.

vidushiy25, 2024-02-03:
My design for now doesn't include the fpu. I have simulated it and it works perfectly fine but gives errors in Quartus. I have set the project within the cv32e40p directory so it has access to all other directories and included all the files in the cv32e40p_manifest.flist in addition to my other soc files. It still gives me syntax errors.

MikeOpenHWGroup, 2024-02-03:
Hi @vidushiy25, can you provide the following:
1. The Operating System and version you are using.
2. The **_exact_** command-lines you are using, starting with `git clone...`.
3. Attachments for any Quartus-specific files you are using (such as configuration file, the top-level instanitation of the core, etc.)

vidushiy25, 2024-02-03:
1. I used the scripts on Linux OS on NoMachine (for university use) for simulation and exported the whole project to my HP laptop which has  Windows 11 so I could synthesise the design using Quartus.
2. To get the files, I just used `git clone https://github.com/openhwgroup/cv32e40p.git`. I then used the example testbench provided in the folders and got it to work. I made my own scripts and example code tests to understand how the core works. Finally, I made an SOC design with the core and a C program to compute factorials of numbers. The design simulates correctly on Modelsim and Xmverilog. 
3. My Quartus project was then set up in the cv32e40p. All files in the .flist were added to the project, and the hex file of the c program was made available in the same directory so that Quartus could access it to load instructions to ROM. The top level module contains a wrapper for de2115 altera fpga that instantiates the soc which comprises of ROM, RAM and an ahb interconnect (Note: The core's obi interface was connected to an obi2ahb adapter and simulated showing correct functionality). I am attaching the de2_wrapper.sv, the cv32e40p_top_ahb.sv, which instantiates the core and the cv32e40p_soc.sv which contains the soc design.  (Note: this soc design is similar to another core whose implementation has been verified on the FPGA. I tweaked the design so as to use the cv32e40p and made sure all files are available for it as mentioned in the docs).
[Archive.zip](https://github.com/openhwgroup/cv32e40p/files/14152613/Archive.zip)

pascalgouedo, 2024-02-05:
Hi,
Just tried analysing & synthetizing all **original** cv32e40p files with cv32e40p_core as top level with quartus prime pro 22.1.0 and everything works just fine without any errors.
[image]

By the way we can see that you modified the RTL source code by adding those timeunit commands.
Better to give that at simulator command rather than touching original RTL files.

pascalgouedo, 2024-02-05:
You can also remove cv32e40p_tracer_pkg from your qsf.

vidushiy25, 2024-02-05:
I just tried the Analysis and Synthesis of the original untouched files from scratch by git cloning the repo into a different folder again and I get the same errors. I am using Quartus Prime Standard edition 23.1std.0. I made the project in the cv32e40p directory and added necessary files. Is there anything different that you did? I'm assuming I don't need the .sdc file until I start compilation.

vidushiy25, 2024-02-05:
I used the cv32e40p_core as the top level entity as well.

pascalgouedo, 2024-02-05:
> I just tried the Analysis and Synthesis of the original untouched files from scratch by git cloning the repo into a different folder again and I get the same errors. I am using Quartus Prime Standard edition 23.1std.0. I made the project in the cv32e40p directory and added necessary files. Is there anything different that you did? I'm assuming I don't need the .sdc file until I start compilation. The target technology I used was the Cyclone IV EP4CE115F29C7.

I made nothing special and didn't use any sdc.
Opened quartus gui, selected files, choose basic Cyclone 10 and ran synthesis.
Only difference is that I have prime pro while you have prime standard but I guess this can't be the source of the pb, this system verilog syntax error...

vidushiy25, 2024-02-05:
I did the exact same but our target technologies are different but I am still confused as to why it isn’t working.

pascalgouedo, 2024-02-05:
By looking on the net it seems standard edition is a quite old quartus code base and prime pro is essentially a re-write of nearly the entire code base.
And [this document](https://github.com/openhwgroup/cv32e40p/files/14164744/quartus-prime-compare-editions-guide.pdf) highlights that standard edition has limited system verilog language support.

pascalgouedo, 2024-02-05:
It is more talking about interface modports than packages import.

I don't like that but move in every files the package import above the module name for the ones included in the module header and let me know if it resolves the pb on your side.

vidushiy25, 2024-02-05:
[image]
It unfortunately doesn't solve the problem. Looks like I might need to use Quartus Prime Pro.

pascalgouedo, 2024-02-05:
This is not the same pb.
At least import pb disappeared.
Here it seems to be caused by genvar j; which is used when PULP_SECURE = 1 which is not our case.
So here we have genvar without for loop using it...

vidushiy25, 2024-02-05:
So does this now have to do with difference of using Quartus Prime Pro vs Standard?

pascalgouedo, 2024-02-05:
Pro seems ok with original code.
Trying to find solutions to overcome Standard version.

Try to move line 287 "genvar j;" in line 1132, just before generate

```
  genvar j;

  generate
    if (PULP_SECURE == 1) begin : gen_pmp_user

      for (j = 0; j < N_PMP_ENTRIES; j++) begin : CS_PMP_CFG

```

vidushiy25, 2024-02-05:
Doesn't change anything..

pascalgouedo, 2024-02-05:
Strange.
Line 289 ?
Standard is very bad.

vidushiy25, 2024-02-05:
I tried. Doesn't seem to work again. This would also mean modifying other files for this. I might consult the person supervising my project to consult other options for using other FPGAs. Thank you for your help!

vidushiy25, 2024-02-13:
The FPGA implementation using Quartus Prime Standard doesn't work, but I can successfully make it work on Nexys4 (Artix-7) using Vivado.

closed 2024-02-13 by vidushiy25

MikeOpenHWGroup, 2024-02-13:
Siemens has a licensing strategy that allows users to control their licensing costs by restricting the HDL language constructs used in their RTL.  For example, ModelSim is a low-cost HDL simulator that does not compile the full IEEE 1800-2017 LRM.  At the OpenHW Group we deliberately exploit the advanced capabilities of languages and tools and that often means low-cost tools cannot support our IP.  In the case of Quartus Prime Standard it is probably a good idea for us to explicitly mention this.

I have created  Issue #948 to track this.  Thanks for working with us on this IP @vidushiy25!