openhwfoundation/cv32e40p #945: Missing files

Amal-k-Ayyan, 2024-02-03:
Following files where found missing:
core_clock_gate
gen_clock_gate
cv32e40p_pmp
cv32e40p_clock_gate

MikeOpenHWGroup, 2024-02-03:
Hi @Amal-k-Ayyan, thanks for your interest in OpenHW cores.  I am afraid I do not understand your comment.  What are these files missing from?  Please make sure you are using the most up to date version of the code.  For example, as of #643, the file `rtl/cv32e40p_pmp.sv` not longer exists.

Amal-k-Ayyan, 2024-02-03:
Hi @MikeOpenHWGroup 
I tried to get the schematic from rtl code given in the repo using vivado. 
But it gave errors showing above mentioned files where missing.

MikeOpenHWGroup, 2024-02-04:
Please ensure you have read the [Core Integration](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#core-integration) section of the User Manual, particularly the [Parameters](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#parameters), [Clock-gating](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#clock-gating-cell) and [Synthesis Guidelines](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#synthesis-guidelines).

If you are still having trouble after that, please provide the _exact_ command-lines you are using, start with `git clone...`.  If there are any files you are using that are not part of this repo, please attach these to the issue.

closed 2024-02-04 by Amal-k-Ayyan