openhwfoundation/cv32e40p #946: Seeking Information on RTL Access

arunvthampi, 2024-02-04:
Hello, I'm new to this Git repository. Is this repository related to the RTL for the CV32E40P core? If so, could you please guide me on obtaining the RTL for RISC-V? Is it necessary to configure any files to obtain the RTL for the core?

vidushiy25, 2024-02-04:
Hi @MikeOpenHWGroup , we're not on the same team or class, but thank you! I have understood my way around the cv32e40p files with the help of the User Manual and experimental testbenches. The Manual has been useful, thanks!

MikeOpenHWGroup, 2024-02-04:
Interesting.  As far as I know, none of the OpenHW Group members use Quartus.  I will check on this...

Which simulator are you using?

vidushiy25, 2024-02-04:
I have used ModelSim and Xmverilog.

arunvthampi, 2024-02-04:
Yes, I am working with Amal K Ayyan. Is it possible to get the seminar

MikeOpenHWGroup, 2024-02-04:
> Is it possible to get the seminar

Reach out to me at mike@openhwgroup.org.

pascalgouedo, 2024-02-05:
> Hi @MikeOpenHWGroup , we're not on the same team or class, but thank you! I have understood my way around the cv32e40p files with the help of the User Manual and experimental testbenches. The Manual has been useful, thanks but I am still facing problems with Quartus. Simulation works perfectly fine but Analysis and Synthesis in Quartus is giving erros.

Just made a try using quartus prime pro 22.1.0 and everything works fine.

MikeOpenHWGroup, 2024-02-13:
Hi @arunvthampi, have your questions been answered?  If so, please close this issue.

closed 2024-02-13 by arunvthampi