openhwfoundation/cv32e40p #947: Incorporating FPU

vidushiy25, 2024-02-08:
Hi, I have some questions regarding incorporating the FPU. I read the documentation but found some bits confusing. When trying to incorporate the FPU do I just set the FPU parameter to 1? The docs say :
 `As mentioned in RISC-V Privileged Architecture specification, mstatus.FS should be set to Initial to be able to use FP instructions. If mstatus.FS = Off (reset value), any instruction that attempts to read or write the Floating-Point state (F registers or F CSRs) will cause an illegal instruction exception.`

So do I need to set mstatus_fs  to Initial somewhere? If yes, where?

vidushiy25, 2024-02-08:
Solved it!

closed 2024-02-08 by vidushiy25

pascalgouedo, 2024-02-08:
Nothing confusing. You don't have to incorporate anything.
As explained you just have to take **all the design files** including the new top level module cv32e40p_top and set FPU to 1.

FS is a field of MSTATUS CSR. Initializing it is at Initial state should be done in the software with a csrw instruction.