openhwfoundation/cv32e40p #958: cv32e40p implementation on  Arty a7 100T FPGA

Amal-k-Ayyan, 2024-03-14:
I would like to try out FPGA synthesis and implementation of cv32e40p core. Is there any constraint file available. If so could anyone please provide it. Also is it possible to use cv32e40p_sim_clock_gating as clock gating as I don't know how to change it based on technology of synthesis. Currently I am planning to implement riscy core on Arty a7 100T. 

Please provide insights on how to do this.

MikeOpenHWGroup, 2024-03-14:
Hi @Amal-k-Ayyan, thanks for your interest in the CV32E40P.   We do not provide support for synthesis into a specific target technology, nor do we have an example system you can use to get the core up-and-running in an FPGA.  Here are a couple of suggestions:
1. Look at the [Synthesis Guidelines](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#synthesis-guidelines) in the User Manual.
2. Have a look at [x-heep](https://github.com/esl-epfl/x-heep).

closed 2024-03-14 by MikeOpenHWGroup

Amal-k-Ayyan, 2024-03-15:
Thanks @MikeOpenHWGroup. 
As I have said before, do I need to change the clock gating or shall I use the same one as I don't have much knowledge in clock gatings. 
Also if I were to make a constraint file for arty a7 100t, should I just the inputs as shown in the photo given below.

[image]

Amal-k-Ayyan, 2024-03-15:
Thanks @MikeOpenHWGroup . Will look into it.
As I have said before, do I need to change the clock gating or shall I use the same one as I don't have much knowledge in clock gatings.
Also if I were to make a constraint file for arty a7 100t, should I just the inputs as shown in the photo given below.
[image]