openhwfoundation/cv32e40p #988: Code coverage report

bekbeis, 2024-05-20:
Hi,

Do you guys have any reports related to coverage of your verification infrastructure? In other words, how good/mature is your verification environment?

Thanks!

MikeOpenHWGroup, 2024-05-20:
Hi @bekbeis, thanks for your interest in the OpenHW Group CV32E40P.  The short answer to your question is "yes".  The long answer is, well, longer.

The OpenHW Group uses [symantic versioning](https://semver.org/) to create releases of our CORE-V IP.  Any major release (e.g. v1.0.0, v2.0.0) can be considered "production quality IP".  Any non-major release (e.g. v1.8.1) cannot be considered ready for silicon.

We do not publish our test and coverage results during on-going development.  Rather, we publish these results upon a major release. We keep these reports in the [core-v-programs](https://github.com/openhwgroup/programs) repository. For the CV32E40P v1.0.0 (released on 2020-12-10) we have published a full release checklist and coverage reports. You can find these [here](https://github.com/openhwgroup/programs/tree/master/Project-Descriptions-and-Plans/CV32E40Pv1/Milestone-data/RTL_Freeze_v1.0.0).

We expect to be releasing the CV32E40P v2.0.0 in the coming weeks.  An initial version of these reports (for v1.8.1) were just published [here](https://github.com/openhwgroup/programs/tree/master/Project-Descriptions-and-Plans/CV32E40Pv2/Milestone-data).  Please be aware that the v1.8.1 reports are not the final reports - we pushed that it to facilitate reviews.

To understand the distinction between CV32E40P v1 and v2, please refer to the [User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/core_versions.html).

[Here](https://docs.google.com/presentation/d/1XrhbHpFRYtAiSvAMPhf7b5oIIrpJKBIv/edit?usp=sharing&ouid=112401338207375688045&rtpof=true&sd=true) is a presentation that briefly introduces our quality goals for creating major releases of our CORE-V IP.

Note that none of the other CORE-V cores currently under development have yet to achieve a formal release.  If you follow OpenHW on social media (Linked-In, etc.), you'll know when we do. :wink: 

If I've answered your question, please close this issue - thanks!

bekbeis, 2024-05-21:
Thank you so much for such an extensive response!

bekbeis, 2024-05-22:
Hi @MikeOpenHWGroup, one more question. 
We are testing some ideas around testbench qualification and we wanted to understand how to run all the tests that bring the coverage to what's in the report. Are there any detailed instructions on how to setup a UVM environment and access all the tests?
Thanks!

MikeOpenHWGroup, 2024-05-22:
@bekbeis, please reach out to me at mike@openhwgroup.org and we can discuss.  (We do not use GitHub Issues as discussion boards.)

closed 2024-05-22 by MikeOpenHWGroup