openhwfoundation/cv32e40p #1001: CVFPU RTL updates for implementation tools

pascalgouedo, 2024-06-12:
Hi @MikeOpenHWGroup & @davideschiavone 

A new CVFPU [PR](https://github.com/openhwgroup/cvfpu/pull/127) related to implementation tools has been opened.
Even if Synopsys Design Compiler didn't complain up to now, it seems Synopsys Fusion Compiler doesn't like those few SystemVerilog RTL lines.
As this is quite an important point to be implementation tools friendly and this PR is updating 4 of CVFPU files, do we need to import those modifications in CV32E40P before v2 RTL Freeze?
If yes, we have 2 solutions:
- create a new CVFPU release including all the updates since v0.8.1 made in August 30, 2023 and vendorize it in CV32E40P repo or
- keep v0.8.1 and create a patch to include this PR modifications.

In both solutions we can make LEC with or without those modifications and create a new RTL tag if LEC is fine for all 7 configurations.
On verification side this new CV32E40P tag could be used on simulation non-regressions but couldn't be used for RISC-V ISA Formal Verification as it would mean killing 19 days jobs (still running!) and launch them again.

Thank you to give your thought about that.

Pascal.

davideschiavone, 2024-06-12:
hi @pascalgouedo - the ideal solution would be revendorize cvfpu - not sure about the point you made about Formal Verification though - I agree it is not nice to kill 19 days.. but then that means we are an older RTL - However, if the new CVFPU is LEC with the old one, then it does not matter to me

pascalgouedo, 2024-06-12:
Thank you for your feedback @davideschiavone .

Finally none of these 2 solutions will be possible before RTL Freeze because it would impact RTL Code Coverage reports, waiver files and even maybe requests a new code coverage gathering from scratch.