openhwfoundation/cv32e40p #1002: Missing area metrics for ASIC synthesis

thomasdingemanse, 2024-06-12:
## Missing area metrics for ASIC synthesis

The [FPGA synthesis](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#asic-synthesis) section of the CV32E40P User Manual says the following about the area:

> The core occupies an area of about XX kGE. With the FPU, the area increases to about XX kGE (XX kGE FPU, XX kGE additional register file).

It seems like `XX` is meant to be a placeholder for specific numbers. I'd be happy to help, but I'm not sure how the core is (supposed to be) synthesized for ASIC to get accurate area metrics. Is there an example workflow I could follow or a synthesis script I could use?

### Component
Component:Doc

### Steps to Reproduce
Not applicable.

pascalgouedo, 2024-06-12:
Hi
Yes [ASIC synthesis](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#asic-synthesis) section contains place-holders for CPU and FPU size numbers in KGE unit. All those sentences will be replaced by new text explaining the synthesis context and some new tables.
We are collecting the numbers right now.

No workflow is available as the Core is available as a soft IP.
An IOs constraint file is given as an example in constraints directory.

closed 2024-07-06 by thomasdingemanse

pascalgouedo, 2024-07-08:
PR #1020