openhwfoundation/cv32e40p #1021: Error during elaboration in xcelium

vasanth00, 2024-07-06:
[image]
 Error during elaboration in xcelium 
xmelab: *E,CUVMUR (./cv32e40p_sleep_unit.sv,154|38): instance 'cv32e40p_top.core_i@cv32e40p_core<module>.sleep_unit_i@cv32e40p_sleep_unit<module>.core_clock_gate_i' of design unit 'cv32e40p_clock_gate' is unresolved in 'worklib.cv32e40p_sleep_unit:sv'.

i can't find out exact reason 
need help

closed 2024-07-07 by vasanth00

chose-o, 2024-10-15:
me too, bro

chose-o, 2024-10-15:
bro, I have solved this problem. You can find these missing document in web: https://github.com/pulp-platform/pulpissimo/tree/master/target, .../pulpissimo-vcu108/rtl for example.