openhwfoundation/cv32e40p #1028: CV32E40P_TRACE_EXECUTION

nimakolahi, 2024-07-16:
I have been trying to enable the core tracer log file to check the instruction execution time, PC, instruction itself, and so on. According to the documentation, I should define CV32E40P_TRACER_EXECUTION in the simulation, and for this reason, I added "'define CV32E40P_TRACER_EXECUTION" at the beginning of the tb_top.sv.

However, I am still not getting the trace_core_<HARTID>.log file.
Can anyone help me with how I can enable the tracer?

pascalgouedo, 2024-07-16:
Hi,
This simple (non-UVM) core test-bench is given as an example and it doesn't instantiate the tracer.

You can have a look in cv32e40p git repo in bhv/cv32e40p_tb_wrapper.sv how it is connected.
By the way the up-to-date tracer aligned with v1.8.3 tag is now enabled with CV32E40P_RVFI and CV32E40P_RVFI_TRACE_EXECUTION.

nimakolahi, 2024-07-16:
> Hi, This simple (non-UVM) core test-bench is given as an example and it doesn't instantiate the tracer.
> 
> You can have a look in cv32e40p git repo in bhv/cv32e40p_tb_wrapper.sv how it is connected. By the way the up-to-date tracer aligned with v1.8.3 tag is now enabled with CV32E40P_RVFI and CV32E40P_RVFI_TRACE_EXECUTION.
> 
> Anyway not sure this behavioural tracer can be added if you want to generate/use a verilator model...

Based on what I see in the cv32e40p_tb_wrapper.sv, I see some conditional 'ifdef s in which I think if we just insert 'define CV32E40P_TRACER_EXECUTION it should instantiate if I am not wrong. (I am not a sverilog expert!) However, then I should simulate the cv32e40p_tb_wrapper.sv, shouldn't I?

pascalgouedo, 2024-07-16:
There are 2 different cv32e40p_tb_wrapper.sv files:
- 1 in core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv without tracer used in simple core tb (core-v-verif/cv32e40p/tb/core/tb_top.sv)
- 1 in cv32e40p/bhv/cv32e40p_tb_wrapper.sv with tracer used in UVM tb (core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv)

pascalgouedo, 2024-07-16:
And the define is CV32E40P_TRACE_EXECUTION, not CV32E40P_TRACER_EXECUTION.

nimakolahi, 2024-07-16:
> There are 2 different cv32e40p_tb_wrapper.sv files:
>
> * 1 in core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv without tracer used in simple core tb (core-v-verif/cv32e40p/tb/core/tb_top.sv)
> * 1 [...]

The CV32E40P_TRACE_EXECUTION is correct. You are right; it was a typo when I wrote it here.
What I have now in my environment is the cv32e40p/bhv/cv32e40p_tb_wrapper.sv and I don't have the core-v-verif.

nimakolahi, 2024-07-17:
Ok great, I started modifying the top_tb.sv in "./example_tb/core/top_tb.sv".
I instantiated the cv32e40p_tb_wrapper and of course I modified the module part of the code accordingly. However, when I compiled the code I had error on $readmemh(firmware, wrapper_i.ram_i.dp_ram_i.mem); line so I commented it.
Considering that I don't know much systemverilog, I keep having several errors such as:
Error (suppressible): ./bhv/cv32e40p_tb_wrapper.sv(157): (vopt-7063) Failed to find 'instr' in hierarchical name 'cv32e40p_top_i.core_i.id_stage_i.instr'
Error (suppressible): ./bhv/cv32e40p_tb_wrapper.sv(164): (vopt-7063) Failed to find 'operand_a_fw_id' in hierarchical name 'cv32e40p_top_i.core_i.id_stage_i.operand_a_fw_id'
I saw that it is suppressible so I thought it worths checking by suppressing. I have some other bugs so I am solving them now.
However, do you think that I can suppress such errors?

pascalgouedo, 2024-07-18:
No.
The connection are needed to correctly extract the information from the core to make the final trace log.
This cv32e40p_tb_wrapper.sv file has been made to correctly bind only in the scope of the core-v-verif UVM test-bench.

It maybe needs some kind of hierarchical level parametrization to allow it to work on both core-v-verif UVM  and simple core test-bench.

Ribisl, 2024-11-20:
So if you don't have access to UVM compatible software you can't use the tracer?
I only have access to Verilator, are there any other options to debug this core? At the moment if I open a wave file with gtkwave I only get the instructions in binary representation, which is obviously hard to debug and further develop. Are there any other options for me?

nimakolahi, 2024-11-20:
> So if you don't have access to UVM compatible software you can't use the tracer? I only have access to Verilator, are there any other options to debug this core? At the moment if I open a wave file [...]

I modified the testbench and printed the instruction and PC values and by parsing the log file of the questasim I kinda made my own tracer. You can do the same alternatively.

Ribisl, 2024-11-20:
@nimakolahi 
Is your modified testbench / repo available for public. So you are using questasim and not verilator right?

nimakolahi, 2024-11-20:
@Ribisl 
Currently, I don't have any public repos. But, yes I am using questasim. I assume you can have the same procedure with verilator as well.

MikeOpenHWGroup, 2024-11-20:
Hi @nimakolahi, it seems you have found a path forward.  As you have seen, the RVFI Tracer used by this core requires a SystemVerilog simulator capable of supporting UVM such as Siemens-EDA Questasim, Synopsys VCS, etc.  There is also a free-to-use SV simulator from Metrics Design Automation called DSim that supports UVM.

Are we able to close this issue now?

nimakolahi, 2025-01-28:
@MikeOpenHWGroup Sorry for the late notice! Yes I found my way and I created my own logic tracer. If you need it I can commit. Anyways, the issue is over.

closed 2025-01-28 by MikeOpenHWGroup