openhwfoundation/cv32e40p #1031: Instruction reencoding documentation

Guilherme-Soares-Sequeira, 2024-07-31:
I am struggling to find detailed documentation of the changes made from the original pulp instructions when re-encoding them for their core-v version.
The most detailed I have found is [this file](https://github.com/pulp-platform/riscv-gnu-toolchain/blob/master/core-v.rst), but as it's not under the openhwgroup repository I am not sure of how up to date / valid it is. Additionally, it appears some instructions never made it to the core-v version, at least in their original form, but that appears to not be documented there.
Thank you

MikeOpenHWGroup, 2024-07-31:
Hi @Guilherme-Soares-Sequeira, I doubt we have what you are looking for.  All of the documentation for this core can be found in the `docs` directory of this repo (and is rendered to HTML using ReadTheDocs ([link](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/))).

Our User Manual captures the instructions supported by the specific versions of the core (e.g CV32E40P v1.8.3).  We do not feel it would be useful to support a "delta-doc" between the versions, although in theory, you could create your own as all the documentation and RTL from all previous tags are available (see the first page of the User Manual for the links).

You are right that some instructions from RI5CY did not make it to the CV32E40P post v1.0.0.  Once again, we do not explicitly document those changes.  You best bet is to use the latest version (currently v1.8.3) and the pre-built toolchain from Embecosm available [here](https://www.embecosm.com/resources/tool-chain-downloads/#corev).

closed 2024-07-31 by MikeOpenHWGroup

Guilherme-Soares-Sequeira, 2024-07-31:
Indeed, that is what I've based my work so far on. The issues arose when trying to re-use the core-v-verif tests, which use the original pulp instructions. Nonetheless, thank you for the response.

MikeOpenHWGroup, 2024-08-01:
Can you create an Issue in the CORE-V-VERIF repo about that?  This should be fixed.  Thanks.