openhwfoundation/cv32e40p #1033: Floating-Point Performance Bug for FPU_*_LAT>1

jenniferhellar, 2024-11-19:
**Component:RTL**

Hi, we've identified a performance bug in the decoder/APU dispatcher.  If FPU_*_LAT>1, the current implementation stalls FPU instructions until the previous completes even if there are no data hazards.

# Steps to Reproduce

## RTL Configuration
COREV_PULP=0, COREV_CLUSTER=0
FPU=1, ZFINX=0
FPU_ADDMUL_LAT=2
FPU_OTHERS_LAT=2

## Software Test
A sequence of independent FMADD (or other FPU) instructions:
```
    878a:	b0001073          	csrw	mcycle,zero
    878e:	3200f073          	csrc	mcountinhibit,1
    8792:	68e7f043          	fmadd.s	ft0,fa5,fa4,fa3
    8796:	68e7f0c3          	fmadd.s	ft1,fa5,fa4,fa3
    879a:	68e7f143          	fmadd.s	ft2,fa5,fa4,fa3
    879e:	68e7f1c3          	fmadd.s	ft3,fa5,fa4,fa3
    87a2:	68e7f243          	fmadd.s	ft4,fa5,fa4,fa3
    87a6:	68e7f2c3          	fmadd.s	ft5,fa5,fa4,fa3
    87aa:	68e7f343          	fmadd.s	ft6,fa5,fa4,fa3
    87ae:	68e7f3c3          	fmadd.s	ft7,fa5,fa4,fa3
    87b2:	68e7fe43          	fmadd.s	ft8,fa5,fa4,fa3
    87b6:	68e7fec3          	fmadd.s	ft9,fa5,fa4,fa3
    87ba:	01d7a027          	fsw	ft9,0(a5)
    87be:	3200e073          	csrs	mcountinhibit,1
```

## Waves
Unnecessary pipeline bubbles:
[image]

For comparison, when FPU_ADDMUL_LAT=1, there are no stalls...
[image]

# Source
The primary (but likely not only) culprit seems to be that apu_lat (the encoding of the instruction latencies) is only 2 bits wide, and for FPU_*_LAT>1, the encoding for normal FPU instructions is the same as the "max latency/always stall" encoding for DIV/SQRT (2'h3).

Decoder...

[image]

Dispatcher...

[image]

# Comments
The pipeline details page of the User Manual outlines the expected data hazard stalls, but doesn't mention this non-data hazard stall at deeper pipeline depths.  This text further down is ambiguous, "Floating-Point instructions are dispatched to the FPU. Following instructions can be executed by the Core as long as they are not FPU ones and there are no Read-After-Write or Write-After-Write data hazard between them and the destination register of the outstanding FPU instruction."  It's unclear if this is saying only that the integer pipeline can execute independent instructions while the FPU is active, or also implying that any further FPU instructions will be stalled. 

Given that the FPU is pipelined, the core RTL should either be fixed to utilize that pipeline or else the documentation updated to clearly state that only one FPU instruction can be inflight at a time.  The latter is critical to know for deeper pipelines since you incur the full latency of each FPU instruction every time unless you can hide enough integer instructions in between which is not usually feasible.

More broadly speaking,

1. Was there any awareness of or discussions around fully utilizing the FPU pipeline – was this intentionally not implemented?  If so, why?
2. Were there discussion around doing performance verification i.e. verifying that the micro-architecture is not only functionally correct, but also correct to the spec in terms of the timing of different sequences of instructions?  If so, what were the factors and conclusions?

pascalgouedo, 2025-03-17:
Hi Jennifer,

Didn't see your issue before, sorry.

I will try to answer with what I recall for that particular performance problem you are seeing.
It is a known problem and the solution to allow multiple multi-cycles FPU instructions when FPU_*_LAT>1 is complex.
Firstly you could have out-of-order responses between DIV/SQRT and other FPU instructions when FPU_*_LAT>1. So you need a request/response ID (which is present on FPU side but not used in the Core) and a re-ordering FIFO/CAM on the Core side to memorize the ID/destination register couple. This would also make much more complex the RAW/WAW detection as well.
But this could be avoided by adding more complex decoding for this stall_type signal to exclude those 2 kind of instructions flowing together in the FPU pipeline.
Secondly I remember that I added some APU result memorization logic to correct many bugs of returning FPU data wrongly forwarded while the core was totally stalled for other reasons (ALU, misaligned LOAD or even Branch).
You can find it around "APU result memorization Register" in cv32e40p_ex_stage.sv (apu_rvalid_q, apu_result_q, apu_flags_q).
This is because apu interface does not have a response grant on the core side so the FPU valid returning data MUST be immediately taken/used even if the core is stalled. There is a response grant on FPU side (out_ready_i), I tried to use it but it didn't work for a reason I don't remember (maybe deadlock or really bad side-effect?).
So if you can't use this response grant, you have to memorize the returning FPU data to be used later once the core is not stalled anymore and there is not conflict on forwarding paths.
It costed some gate count but there was no other solution I could find to resolve all those bugs.
Allowing more than 1 multi-cycles instruction to enter in the FPU pipeline when their latency is greater than 1 would mean to replace this register bank by a real FIFO (2-3 elements?) for data plus another specific one to keep destination registers when requests are granted. And adding more complex RAW/WAW decoding as well (timing critical?). The additional cost seemed too high for a small core like CV32E40P so it stayed like that, meaning only 1 multi-cycles instruction at a time in FPU pipeline.

Moreover even if those lost cycles can not be totally avoided, you can still mask some of them with a good compiler or assembly programmer by unrolling loops or by using other tricks.

Regards,
Pascal.

jenniferhellar, 2025-03-25:
Hi Pascal, thanks for the detailed response.  Given that, I would strongly recommend/request:

1. Updating the documentation for this release to explicitly state that regardless of FPU configuration, the core utilizes the FPU as if it is not pipelined, allowing only one FPU instruction inflight at a time, and therefore the full *_LAT cycles is always incurred for sequential FPU instructions.  It's a major design/performance limitation that's not at all obvious presently.

2. Creating some sort of linked future feature/future optimization tracker (perhaps this thread) in case someone who really cares about FPU performance wants to fix this down the road.  They would have a starting point to understand the design decisions made and issues to solve.

All the best,
Jennifer

MikeOpenHWGroup, 2025-03-26:
Hi @jenniferhellar and @pascalgouedo. To Jennifer's points:

1. I would recommend making this clarification in the `Cycle Counts per Instruction Type` table in the [pipeline](https://github.com/openhwgroup/cv32e40p/blob/360d272898d81806be3377193870dbf83a3ea79f/docs/source/pipeline.rst?plain=1#L146) chapter.  Knowing what I know now (after reading this issue), it could be said that the table already hints at that, but Jennifer is right, it could be more clear.  Unless there are any objections, I will generate a pull-request to add Jennifer's comments to that table.

2. OpenHW does not have a good process for raising, tracking and discussing "future feature/future optimizations".   As an experiment, I have created a Discussion Board on this repo. and in a future pull-request I will mention it in the top-level README. @jenniferhellar, you will forever be known as an OpenHW Trailblazer if you would add this feature to that discussion!

pascalgouedo, 2025-03-26:
Hi Jennifer,

> Updating the documentation for this release to explicitly state that regardless of FPU configuration, the core utilizes the FPU as if it is not pipelined, allowing only one FPU instruction inflight [...]

This is not that bad!
FPU is pipelined and the core can send more that 1 FPU instruction in its pipeline when LAT = 1 (0 means not pipelined FPU).
So the pb you have seen only occurs when LAT is greater than 1 and we cannot make the difference between real 3 cycles FPU intructions and DIV/SQRT (plus the pb  of multiple returning FPU results).

BTW having LAT = 1 is generally sufficient to achieve high frequency when using the FPU close to the core like integrated in cv32e40p_top.
We used LAT = 2 in a multi-core cluster context (originally pulp_cluster from Pulp Platform) because one FPU was shared between 2 cores with some muxing before/after it.

pascalgouedo, 2025-03-26:
> I would recommend making this clarification in the Cycle Counts per Instruction Type table in the [...]

I agree that the description for the 3 last lines of this cycle count table (Table 63) is not that clear and easily understandable, my bad.
The text is correct for DIV/SQRT.
But for the 2 lines concerning ADD/MUL and OTHERS FPU instructions, a different text could detail the cycle count for the 0, 1 and 2 cycles latency cases to avoid any misinterpretation.
I could think about it and propose something during my spare time.

MikeOpenHWGroup, 2025-03-26:
> I could think about it and propose something during my spare time.

Gracias mi amigo!

jenniferhellar, 2025-03-27:
@pascalgouedo it depends on the technology node.  On our tech, we need ADDMUL_LAT=2 to reach even moderate frequency and ADDMUL_LAT=3 to reach high frequency.  With register retiming, etc.

For programs with low FPU intensity, the performance loss is not huge, but at ADDMUL_LAT=3, on DSP code (IIR, FIR, FFT), this pipelining issue causes **>40-50% performance loss**.  And that is with partially unrolled loops, etc.  I don't have the numbers on hand for ADDMUL_LAT=2, but it is significant for that configuration as well.

The OpenHW primary goal is verification, not design, so I understand why this wasn't fixed at the time.  It would take a non-trivial design effort.  But it is a big enough feature (and easy to miss) to warrant documentation and a discussion about fixing in future core iterations.

Regardless, I very much appreciate your engagement when I know the project is long over and you've moved on!  Thanks

@MikeOpenHWGroup Sure I can put something on the discussion board when I get some time.  I'll probably just write something short and link to this issue.  Thanks for setting that up.

pascalgouedo, 2025-03-27:
Hi Jennifer,

Never heard about anybody using more than 2 cycles for FPU latency numbers.

Yes OpenHW primary goal is verification but I was not an OpenHW employee, I work in industry companies and best performances and best PPA has always been in my blood for more than 30 years working on defining and designing DSPs and CPUs.
I can assure you that if there was any problem like the one you have to reach between 500 and 600 MHz with 2 cycles-latency (that we achieved in a 16 CV32 + 8 FPUs cluster in a GF22FDSOI chip), I would have implemented anything to improve that.
For sure we didn't try that on a large range of technologies but I am still quite surprised that 3 cycles latency is needed for just a Single-Precision Floating Point unit. Except if you use old/slow technology or try to reach GHz range...
Anyway I agree that with some effort this (any) pb could be fixed.

BTW CV32E40Pv2 (v1.8.3 in fact) has been verified with formal ISA tool and simulation verification only for FPU latency values equal to 0, 1 or 2.
But with the stall_type signal you highlighted above, we can see than any LAT value greater or equal than 2 acts the same way from the core point of view so I don't think any bug could appear with LAT value greater or equal than 3.

As you said I moved on to new horizon and I will let you continue to manage this issue even if I am always happy to give any information about what we have done on architecture, design and verification of CV32E40Pv2.

Best regards,
Pascal.