openhwfoundation/cv32e40p #1035: A question About this core on Litex

DaviLdM, 2025-01-13:
I'm using latex to create a SOC for my processor, the cv32e40p can be used in the litex, when I use the command litex_sim --cputype cv32e40p the verilator works normally, however when I use the litex itself for the FPG terasic-de1soc The code presents synthesis errors for the following parts:
>Error (10644): Verilog HDL error at cv32e40p_register_file_ff.sv(142): this block requires a name File: /home/xmen/.local/lib/python3.8/site-packages/pythondata_cpu_cv32e40p/system_verilog/rtl/cv32e40p_register_file_ff.sv Line: 142
Error (10112): Ignored design unit "cv32e40p_register_file" at cv32e40p_register_file_ff.sv(30) due to previous errors File: /home/xmen/.local/lib/python3.8/site-packages/pythondata_cpu_cv32e40p/system_verilog/rtl/cv32e40p_register_file_ff.sv Line: 30
>Error (10170): Verilog HDL syntax error at cv32e40p_cs_registers.sv(326) near text: "csr_rdata_int";  expecting "assert". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/xmen/.local/lib/python3.8/site-packages/pythondata_cpu_cv32e40p/system_verilog/rtl/cv32e40p_cs_registers.sv Line: 326
>Error (10170): Verilog HDL syntax error at cv32e40p_cs_registers.sv(297) near text: "}";  expecting ";". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/xmen/.local/lib/python3.8/site-packages/pythondata_cpu_cv32e40p/system_verilog/rtl/cv32e40p_cs_registers.sv Line: 297

I would like to know why in the verilator simulator the code works and in Litex directly it doesn't

MikeOpenHWGroup, 2025-01-13:
Hi @DaviLdM, thanks for your interest in OpenHW and the CV32E40P.

In theory, it should be possible to support the CV32E40P in the Litex environment.  However, Litex is not an OpenHW project and we do not have the resources to directly support our cores in the Litex environment. The one (possible) exception to this is the [CVA5](https://github.com/openhwgroup/cva5).  That work was contributed about 2.5 years ago and the current status of "CVA5 in Litex" is unknow. Nevertheless I can offer the following comments.

It appears that you are trying to compile the CV32E40P into an Intel/Altera FPGA.  To the best of my knowledge, this has not been attempted by any OpenHW members.  As such, it is not surprising that the Altera tools cannot handle the CV32E40P - it is a very common experience that "proper SystemVerilog code" is correctly processed by one tool, yet not by another.  At OpenHW we do not have access to Altera FPGA tools, so you will need to address this yourself.  Should you wish to contribute a fix, we would be happy to review your pull-request.

closed 2025-01-13 by MikeOpenHWGroup

DaviLdM, 2025-01-13:
Thank you very much for the answer, I will continue trying to solve the problem, and if I succeed, I will send a pull-request.