openhwfoundation/cv32e40p #1038: Incorrect branch target address calculation in specific instruction sequences

nexgru, 2025-03-16:
### Issue Description
Hi, I encountered an unexpected discrepancy during simulation in ModelSim. That is, branch instruction may incorrectly calculate the branch target address in certain instruction sequences.

### Component
**Component: RTL**

### Steps to Reproduce
As shown below, the following sequence of instructions happen:

> 0023006F, 00700567, FFA3FFEF, 0112A167, 10038163, F9C80063

The last instruction F9C80063 is a branch instruction. Its function is: "If the values of general-purpose registers 16 (x16) and 28 (x28) are equal, jump to the offset address FFFFF780."
According to the diagram, the current address of this instruction (if_stage.pc_if_o) is 00000008, so the expected jump target address (id_stage.jump_target_o) should be FFFFF788 (calculated as 0x00000008 + 0xFFFFF780). 

[image]

However, the actual result shows the jump target address as 00000006. This may be a miscalculation.

[image]

### Top Level Parameters
```
cv32e40p_core #(
    .FPU                      ( 1 ),
    .NUM_MHPMCOUNTERS         ( 1 ),
    .COREV_CLUSTER            ( 0 ),
    .COREV_PULP               ( 0 ),
    .ZFINX                    ( 1 )
) 
```

### Test Bench
```
initial
  begin
	clk_i = 1'b0;  
	rst_ni = 1'b0;

	#30
	rst_ni = 1'b1;

	#60
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b00000000000000000000000000000000;
	data_rvalid_i = 1'b1;
	debug_req_i = 1'b0;
	dm_exception_addr_i = 32'b11111111111111111111111111111111;
	dm_halt_addr_i = 32'b11111111111111111111111111111111;
	fetch_enable_i = 1'b1;
	hart_id_i = 32'b00000000000000000000000000000000;
	instr_gnt_i = 1'b1;
	instr_rdata_i = 32'b00000000001000110000000001101111;
	instr_rvalid_i = 1'b0;
	irq_i = 32'b00000000000000001111011101110111;
	mtvec_addr_i = 32'b11111111111111111111111111111111;
	scan_cg_en_i = 1'b0;
	apu_flags_i = 5'b00000;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;
	boot_addr_i = 32'b11111111111111111111111111111111;

	#200
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b00000000000000000000000000000000;
	data_rvalid_i = 1'b1;
	debug_req_i = 1'b0;
	dm_exception_addr_i = 32'b11111111111111111111111111111111;
	dm_halt_addr_i = 32'b11111111111111111111111111111111;
	fetch_enable_i = 1'b0;
	hart_id_i = 32'b00000000000000000000000000000000;
	instr_gnt_i = 1'b1;
	instr_rdata_i = 32'b00000000011100000000010101100111;
	instr_rvalid_i = 1'b1;
	irq_i = 32'b00000000000000001111011101110111;
	mtvec_addr_i = 32'b11111111111111111111111111111111;
	scan_cg_en_i = 1'b1;
	apu_flags_i = 5'b11111;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;
	boot_addr_i = 32'b11111111111111111111111111111111;

	#200
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b00000000000000000000000000000000;
	data_rvalid_i = 1'b0;
	debug_req_i = 1'b0;
	dm_exception_addr_i = 32'b00000000000000000000000000000000;
	dm_halt_addr_i = 32'b00000000000000000000000000000000;
	fetch_enable_i = 1'b0;
	hart_id_i = 32'b00000000000000000000000000000100;
	instr_gnt_i = 1'b0;
	instr_rdata_i = 32'b11111111101000111111111111101111;
	instr_rvalid_i = 1'b1;
	irq_i = 32'b11111111111111111111111111111111;
	mtvec_addr_i = 32'b00000000000000000000000000000000;
	scan_cg_en_i = 1'b1;
	apu_flags_i = 5'b11111;
	apu_gnt_i = 1'b0;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;
	boot_addr_i = 32'b00000000000000000000000000000100;

	#200
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b11111111111111110000000000000000;
	data_rvalid_i = 1'b0;
	debug_req_i = 1'b0;
	dm_exception_addr_i = 32'b11111111111111111111111111111111;
	dm_halt_addr_i = 32'b11111111111111111111111111111111;
	fetch_enable_i = 1'b0;
	hart_id_i = 32'b00000000000000000000000000000000;
	instr_rdata_i = 32'b00000001000100101010000101100111;
	instr_rvalid_i = 1'b1;
	irq_i = 32'b11111111111111111111111111111111;
	mtvec_addr_i = 32'b11111111111111111111111111111111;
	scan_cg_en_i = 1'b0;
	apu_flags_i = 5'b00000;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;
	boot_addr_i = 32'b00000000000000000000000000000011;

	#200
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b11111111111111110000000100000101;
	data_rvalid_i = 1'b0;
	debug_req_i = 1'b0;
	dm_exception_addr_i = 32'b11111111111111111111111111111111;
	dm_halt_addr_i = 32'b11111111111111111111111111111111;
	fetch_enable_i = 1'b1;
	hart_id_i = 32'b00000000000000000000000000000000;
	instr_rdata_i = 32'b00010000000000111000000101100011;
	instr_rvalid_i = 1'b1;
	irq_i = 32'b00000000000000001111011101110111;
	mtvec_addr_i = 32'b00000000000000000000000011111111;
	scan_cg_en_i = 1'b1;
	apu_flags_i = 5'b00000;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;
	boot_addr_i = 32'b00000000000000000000000000001011;

	#200
	data_gnt_i = 1'b1;
	data_rdata_i = 32'b00000000000000000000000000000000;
	data_rvalid_i = 1'b1;
	debug_req_i = 1'b0;
	hart_id_i = 32'b00000000000000000000000000000000;
	instr_rdata_i = 32'b11111001110010000000000001100011;
	instr_rvalid_i = 1'b1;
	irq_i = 32'b00000000000000001111011101110111;
	mtvec_addr_i = 32'b11111111111111111111111111111111;
	scan_cg_en_i = 1'b1;
	apu_flags_i = 5'b00000;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000000;
	apu_rvalid_i = 1'b0;

	#200
	data_gnt_i = 1'b0;
	data_rdata_i = 32'b00000000000000000000000000000100;
	data_rvalid_i = 1'b1;
	debug_req_i = 1'b0;
	hart_id_i = 32'b00000000000000000000000000000000;
	apu_gnt_i = 1'b1;
	apu_result_i = 32'b00000000000000000000000000000100;
	apu_rvalid_i = 1'b0;

	#10000
	rst_ni = 1'b0;
	
	#100
	$stop;
  end
always #100 clk_i = ~clk_i;
```

Product: ModelSim
App: Simulate
Tool's version: ModelSim SE-64 2020.4,  Revision: 2020.10

Clock file, Test Bench and VCD file are provided in the attachments. 

[attachments.zip](https://github.com/user-attachments/files/19275114/attachments.zip)

closed 2025-03-17 by MikeOpenHWGroup

MikeOpenHWGroup, 2025-03-31:
Hi @nexgru, there is no "expected value" for internal signals.  We verify the behavior of the core and do not check internal signals unless their behavior is specified in the [User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/).

nexgru, 2025-04-01:
Hi @MikeOpenHWGroup, thanks for your reply. I'll go check the user manual and CORE-V-VERIF guide.